opentx/opentx #5692: GV values do not transfer, are recast as percentages

sande005, 2018-02-13:
OpenTX 2.2.1.  Verified in Companion and on Taranis x9d+

Units for a GV left blank.  When used as a variable for Mixing weight or as a Min in Outputs, it is treated as a percentage, even though profile is set to use ppmus.  In example attached:  GV set to -90.  Throttle output center is 1500us.  Expected ppmus for low throttle would be 1500-90, or 1410.  Instead it is 1039 ppmus, which is -90% of default 512.

[GV as Percents.zip](https://github.com/opentx/opentx/files/1718618/GV.as.Percents.zip)

3djc, 2018-02-13:
Unless I misunderstood something in your description, it is perfectly normal : weight IS a percentage function (if you do not enter a GV but a plain value,, you will have the very same behaviour). If you want to offset a value, then you the offset function.

sande005, 2018-02-13:
Same thing happens when assigning a GV to an output Min., even though outputs are configured to be in ppmus, The GV always presents as a percentage, so in the above example, if the GV was -90 for Min, one would expect it to be the equivalent of typing -90 in the Min box.  But no, it actually presents as -461 (which is 90% of the default 512 ppmus).  To get the Min to be -90 ppmus, the GV has to be 18 (slightly off, due to rounding of percents).

3djc, 2018-02-13:
Absolutely, % is an information field to help people with a lot of GV to remember how they are used. It does NOT change function behaviour. Like weight, output min and max are % functions

3djc, 2018-02-13:
Ok, I think I understand what you mean. Is that:

[image]

[image]

[image]

kilrah, 2018-02-13:
Did you change the GVAR Precision? That's what does the scaling, not the % choice which is only an indication.

sande005, 2018-02-13:
Yes 3djc, that is a good example of the issue

kilrah - No change after changing the GVAR Precision to include a decimal.  Still treated as a percent, rather than a decimal number.

Maybe it is time for me to get "modern" and just display percentages everywhere.  Kinda like substituting "channels" for "frequencies" back in the 72mhz days.  Most radios use % to be more user friendly.

3djc, 2018-02-16:
I have investigated a little, and here is the issue : in a PPMUS firmware, there is a difference between the value stored, and the value displayed in several places, including outputs. So you see -90 like I have it setup for CH2 for exemple, but the value used for calculation is the % equivalent (a value near -17,6%).

When you put GV1 in that output, it is the real value of GV1 that is used, in my example -90%.

closed 2019-11-15 by stale[bot]