opentx/opentx #5696: GV limits wrong when trim is reused

jfrickmann, 2018-02-17:
When I use a special function to adjust a GV with a trim button, then limits for the trim are applied, where limits for the GV should  be applied instead.
I have attached a copy of opentx.cpp where I have fixed the issue. I haven't yet figured out how to properly commit a patch - sorry!!

Jesper
[opentx.txt](https://github.com/opentx/opentx/files/1733385/opentx.txt)

3djc, 2018-02-17:
Is your special function set to copy trim value, or to inc/dec on trim ?

jfrickmann, 2018-02-17:
It is set to inc/dec on trim. I have added a test OTX above for you.

I have also updated the other files, as there were a couple of errors in the first one.

Jesper
[screenshot from 2018-02-17 06-41-38](https://user-images.githubusercontent.com/16344987/36340556-cb3def98-13ad-11e8-9c67-d3a2faa817a1.png)

3djc, 2018-02-17:
ok, you are copying trim value, that would be incrementing

[image]

jfrickmann, 2018-02-17:
I created my own branch and added a commit to it - is that the "proper" procedure for doing it? 

Thanks!

3djc, 2018-02-17:
Yes, I know, I added the screenshot because you said you where incc/dec, but you are not, you are copying trim value. The screenshot would be inc/dec, that's all

You can ask commit from your branch to opentx:2.2, that would make a PR  from your repo to ours (like for example #5695)

jfrickmann, 2018-02-17:
I managed to make a pull request to have my branch merged in, but the test build failed for AVR9X:

opentx.cpp: In function ‘uint8_t checkTrim(event_t)’:
opentx.cpp:1376:22: error: ‘GVAR_MIN’ was not declared in this scope
opentx.cpp:1376:53: error: request for member ‘min’ in ‘g_model.ModelData::gvars[((int)gvar)]’, which is of non-class type ‘gvar_t {aka int}’
opentx.cpp:1377:22: error: ‘GVAR_MAX’ was not declared in this scope
opentx.cpp:1377:53: error: request for member ‘max’ in ‘g_model.ModelData::gvars[((int)gvar)]’, which is of non-class type ‘gvar_t {aka int}’

This is all inside of a #if defined(GVARS) block, so it is pretty confusing to me. 

Do I need to fix it and make another pull request, or will someone "in the knows" be willing to take a look at it?

Thanks!
Jesper

3djc, 2018-02-17:
It would be great if you can fix it, you do not need to issue another PR, just commit the fix in your branch, it will synchronise automatically (and welcome to the wonderful world of AVR !!!)

kilrah, 2018-02-17:
AVR supports GVARs, but only in their basic form without the extended attributes (min/max/unit/prec).

jfrickmann, 2018-02-17:
OK, I inserted an #if defined(CPUARM), and for #else, I set the limits to TRIM_MIN/MAX as before. I don't know if those are the correct limits for non-ARM platforms with GVARS, so I added a TODO comment to flag the hack. The pull request passed.

Jesper

PR #5698 (2018-04-25): Issue #5696
When an "Adjust GV" special function is reusing trim buttons to increase/decrease GVs, then the trim limits are applied instead of GV limits. This patch applies the GV limits.

This fixes #5696
closed 2018-04-25 by bsongis