opentx/opentx #5697: Feature Request: Multi-protocol module in Turnigy 9XR with m128 chip

kebes22, 2018-02-17:
I have a Taranis X9D+, and a Turnigy 9XR (non-pro, with m128 chip). I am using the X9D+ for myself, and my son is using the 9XR. I have 2 of the iRangeX IRX4 multi-protocol modules (one for each radio). The problem is that I can only enable it in the firmware for the X9D+ (in OpenTX Companion). I have done a little research, and have found that the 9XR should be capable of running the multi-protocol module (possibly with a slight hardware mod), but there does not seem to be the option in OpenTX Companion to enable this for the 9XR. The 9XR Pro does have this option (I have set it up for a friend), but not the 9XR.

I believe er9x supports it, but I would like to keep OpenTX on both radios to make it easier to share models between them, and because honestly OpenTX just rocks. Is it possible to add this as an option for the 9XR. I am guessing that the limited processor on the 9XR may be a problem, but it seems it should be able to maybe disable something else that I don't need to make room?

Is there any way this can be added, or is there some way I can already get it to work with OpenTX on my 9XR?

Thanks.

schwabe, 2018-02-17:
Duplicate of #3812.

Yes it is possible but none of the current OpenTX developers even has a 9X/9XR (or any other AVR radio). And I myself have no desire to develop the code for it.

So unless someone in the community steps up and develops multi support of the AVR radios, the support will not happen.

closed 2018-02-17 by schwabe

powercycle, 2018-02-17:
I thought there was support for that radio and multi protocol around 2.0.19?  So I believe there is support for that for the 9XR, but you can't upgrade above 2.1.

I don't have that radio anymore, but I believe that it was working.

It has been a while, so I could be mistaken.

kilrah, 2018-02-17:
No, never been there. Er9x does support it so that’s probably what you remember.

schwabe, 2018-02-18:
No, there was an unofficial build from me for latest two 2.1.x version but those builds were also x9d+/x9e only.

kilrah, 2018-02-18:
Which is why I said it's never existed for the 9XR as he says ;)