opentx/opentx #5699: bulid opentx  problem

skyyuzhang, 2018-02-18:
everytime when I bulit opentx firmware .the QT always show me  some message which is
 No rule to make target 'D:/opentx-2.2/radio/src/translations/cz.h', needed by 'radio/src/translations/CMakeFiles/firmware_translations'.  Stop.   

somebody could help me?  please

mpaperno, 2018-02-20:
The step which generates the nn.h translation files must be failing. Maybe your Python installation wasn't detected.

closed 2018-02-20 by mpaperno

blockx, 2018-04-09:
I have the same issue with the exact same error message.  If I run the command "python --version" in the CMD window I am running "make" from I get the response "Python 2.7.14" so I think python is been found OK.  Any ideas ?

Mike

mpaperno, 2018-04-10:
When you run CMake to generate build config it shows a bunch of messages, one of which is the Python version which was detected, or a warning message if Python wasn't found.  Check that output.  (If you're using QtCreator to run CMake, the output it shown in the General Messages pane.)  Eg:

```
OpenTX 2.2.2 @eb1aa518
Python found, version: 2.7.9
Qt Version: 5.7.1
```