opentx/opentx #5700: Horus X12S problem with Multimodule

egitter-vc, 2018-02-18:
My Horus X12S is running openTX 2.2.1. MultiMoule is IRX4 PLUS from IRANGEX (the one with the rotary switch).
When trying to config the IRANGEX-Module the as the external module I can not config the external module as a multimodule with the apropriate subtypes on the transmitter. The item is missing. Works on Horus X10S.

Setting it up in Companion, I can configure the external Module as MultiModule with for example DSM DSM2 22ms.
Downloading it to the transmitter works, but does shows not the correct subitems. Trying to configure does not work.
Trying to initiate a bind to a DSM2 reciever does not work. the IPEX4 does not light up any LEDs after switching on the transmitter ans pressing the bind-button

The very same module and setup works on a Horus X10S perfectly.

kilrah, 2018-02-18:
You did not select the "multimodule" firmware option. Select, download and reflash.

closed 2018-02-18 by kilrah

egitter-vc, 2018-02-19:
Solved, exactly the problem. 
ToDos: 
1) tick Multimodule in setup page of companion, 
2) download the appropiate file
3) rename it according to file-naming restrictions 
4) flash it to the transmitter

the IRangeX IPX4 Plus will bind to DSM and DSMX recievers