opentx/opentx #5701: Horus X12S chaning throttle input and reversing it does not work correct

egitter-vc, 2018-02-18:
Using a stick to control throttle works. Inverting throttle and starting a timer with the throttle stick works.

Now the problem: I am moving the throttle-input to a diffrent input (e.g. Poti LS) due to the fact that I want the throttle-stick for butterfly-input. Due to habit I need to reverse LS. 

The alerts for throttle "0" are not working correctly. Trying to invert the LS via throttle-reverse reverses my buttlerfly-input, not the LS poti. This is not correct. Inverting the input-channel works for the Channel but has no effect for the Thottle Alerts and the Starting Timers.
Also starting a timer with throttle via GSs with the inverted setup does not. This is not correct.

kilrah, 2018-02-18:
Such throttle warning customization is currently not supported. 
Duplicate of #2859

3djc, 2018-02-19:
Could you post your otx file, there are elements of this that may show another issue.

egitter-vc, 2018-02-19:
the models name is ElDorado V0
the throttle is on input LS. Pls let me know if you need the pic, soundfiles or anything else.

It's basically a V-tail, 4 flap glider with a motor.
[issue 5701.zip](https://github.com/opentx/opentx/files/1737511/issue.5701.zip)

closed 2019-11-15 by stale[bot]