opentx/opentx #5705: Display of TOT in STATS screen wrong sometimes. X9D+ OTx 2.2.1

mpjf01, 2018-02-20:
OpenTx 2.2.1 cca71808. Transmitter X9D+.

The display of TOT (total time) in the Statistics/STATS screen does not seem to cope with (presumably) large numbers correctly. My radio shows 21E38:02. Not sure if it's an E or just three horizontal lines in front of a 3 (or maybe a B).

kilrah, 2018-02-20:
Can you post a copy of your eeprom so whe can check the actual value?

mpjf01, 2018-02-20:
Sorry but I reset it yesterday. I assume that it would have been a large number as the Tx has been in use for years and I've not reset it during that time. I have attached my most recent backup, that may help.
[Taranis plus backup 9 feb 18.otx.txt](https://github.com/opentx/opentx/files/1742000/Taranis.plus.backup.9.feb.18.otx.txt)

PR #5715 (2018-03-04): Allow more the 2 hours digits for TIMEHOUR
This fixes #5705
closed 2018-03-04 by kilrah