opentx/opentx #5709: LS not properly udpated in the AND switch field

LapinFou, 2018-02-21:
When editing a model, the available LS in the "AND switch" list is not dynamical updated.
It will work once, then we must close the editing windows and reopened it in order the list to update.
Here is a snapshot of the issue:
[image]

The LS01 has been added, but not LS02.

LapinFou, 2018-02-21:
[image]

mpaperno, 2018-02-21:
Which version are you testing,  2.2.1?  Should be fixed in `2.2` branch already, which would also make this something like a "quintuplicate" issue on this subject...  (also please don't assign things to me.)

LapinFou, 2018-02-22:
Oups!! I forgot to mention the version → Companion 2.2.1, Dec 17 2017 Windows version.
I will try on the latest nightly build.
(And sorry fir the "assign". I won't do it again).

LapinFou, 2018-02-22:
I checked the 2.2.1 MacOS → same issue.
I checked the 2.2.2N526 → issue is gone.
So, everything is OK.
Sorry for opening this ticket.

closed 2018-02-22 by LapinFou

mpaperno, 2018-02-22:
No worries, & thanks for confirming the fix.

LapinFou, 2018-02-22:
You're welcome. 😄