opentx/opentx #5710: Sticky logical switch result on TR testing

pindeslandes, 2018-02-21:
[Taranis X9D+ running OpenTx 2.2.1]

Configuration : X9D as trainer, bound to a XM receiver in the battery bay of the X9D+ as master.

In trainer’s transmitter:
- CH15 = -100 (source = MAX, Ratio = -100)
- Failsafe of CH15 is set to 0.

In master’s transmitter:
- LS50 set as a~x, V1=TR15, V2=-100
- CH19 = TR15 (source = TR15, Ratio = 100) to visualize TR15

Sequence:
1) I turn on master’s transmitter: LS50 = OFF, that’s OK
2) I turn on trainer’s transmitter: on master’s transmitter, LS50 turns ON and CH19 = -100, that’s OK
3) I turn off trainer’s transmitter: **LS50 sticks to ON**, that’s not OK, although CH19 = 0 (meaning TR15 = 0) as expected
4) I modify LS50 with V1=CH19 instead of V1=TR15: LS50 turns OFF, as expected

3djc, 2018-02-21:
You do not describe everything, but do you get trainer lost signal ?

pindeslandes, 2018-02-21:
Yes, everything else seems to work fine.

3djc, 2018-02-21:
Then you need to understand that TR channels are not regular channels, their mixer lines are disabled when trainer is lost (or not present).

Your CH19 goes to 0 because it is its default value, not because TR15 is at 0 (it is not available per say)

pindeslandes, 2018-02-21:
OK, I understand, but that doesn't solve the issue: why does LS50 stick to ON?

By the way: how does the master's transmitter know that the trainer is lost if the falesafe has been set pour the XM receiver in its battery bay? Or, put it in another way: what could I use to detect a trainer signal lost?

hrenz, 2018-02-21:
this is my problem too
we got a voice alarm but have not a bit setting for  lost trainer signal for a LS or SF

kilrah, 2018-02-21:
Failsafe positions are irrelevant on sbus. When signal is lost the sbus failsafe flag is set, and any frame that has it set is discarded. So your LS is still true.

3djc, 2018-02-22:
You could expand on your logic with the info you have now...

You take an unused trainer channel (lets take the TR15 of your exemple) an unused master channel (let's take your ch19), and you do

[image]

This way, when the TR15 exist and is sending '0', CH19 is at value -100 ((0 x 100%) - 100), when not, it goes back to it's default value of 0.

But maybe you need all 16 trainer channel ? Then you set the weight to 0 so that you have (TR15 x 0) -100, so even if TR15 is moving, you get -100 when the trainer is active, 0 when not

closed 2018-02-22 by 3djc

pindeslandes, 2018-02-22:
Yes, thank you @3djc .
Then there is another problem:
If I want to hold some TR when trainer signal is lost (for instance I have a few LSs that are triggered on TR variations so I get multiple irrelevant alarms when transmitters are too close one to the other), failsafe can't be used. And I can't see any simple workaround to do so at the moment. I will try to use free GVs, assuming a GV is holding its value when its source is lost, but it would be quite a limited solution.
Maybe the true issue here is what @kilrah describes: _**When signal is lost the sbus failsafe flag is set, and any frame that has it set is discarded.**_

3djc, 2018-02-22:
There is not way to 'hold' channel data (nor should there be one IMHO). When data is not available (that be chanel or sensor), it is not available, period. Remember that the mixer is permanently runinng and computing.

Also, you have some latitude : if you want some LS to stay, well, they will if they are based on TR channel when they disapear

pindeslandes, 2018-02-24:
I checked: GVs hold their value when their source is lost.
Anyway the fact that failsafe positions are irrelevant on Sbus sounds strange to me as the XM (Sbus only) receiver manual does describe how to set failsafe. And according to this manual, if failsafe is not set, it defaults to ‘Hold’.
So I guess this discarding failsafe flag doesn’t belong to Sbus protocol but is an OpenTx issue, right?

3djc, 2018-02-24:
Sure, GV hold the value, so do LS relying on it. When trainer is lost, it basicaly stop existing, so no more computation is done on those values.

Yes SBUS allow you to set failsafe for when you 'speak' to a SBUS servo for example, but when you talk to a more evolved partie (like a FC, or redundancy box, or OpenTX, then the failsafe flag, also part of the protocol, is used. That's how we know trainer signal is lost, and therefore stop use it.

kilrah, 2018-02-24:
> So I guess this discarding failsafe flag doesn’t belong to Sbus protocol but is an OpenTx issue, right?

It is standard behavior for all "intelligent" devices (flight controllers, stabilizers etc) that are capable of doing things better then what a failsafe allows for.
On a SBUS->PWM converter the failsafe positions would be used since the converter is "dumb".

pindeslandes, 2018-02-24:
Thanks to both of you for those explanations.
So wouldn’t it be a good idea that OpenTx allows the user to decide what to do with the failsafe in case the trainer signal is lost, instead of killing it?