opentx/opentx #5711: Simplify code by converting some or all data structs to classes

jfrickmann, 2018-02-22:
I believe that the code could be made simpler and easier to understand by converting the structs in datastructs.h to classes with their own methods for accessing and manipulating data.
As long as we do not add any virtual methods (adding a vtable pointer), the underlying data structures will not change. But the code and the compiler directive circus could be kept in one place, instead of being scattered across the different code components using the data structs.
I suggest that we add a new file named datastructs.cpp to /radio/src for the code. Then I would take an incremental approach, where one struct at a time is converted, and the code using it is modified to use the member functions.
I will start working in my own branch, and submit pull requests when I have something to share, if you don't mind?

Jesper

kilrah, 2018-02-22:
There is already extensive storage refactoring in the works. 
The current code is as it is becasue on atmega64 platforms every byte of code memory is precious so efficient compiler output takes precedence over source cleanliness or readability.

jfrickmann, 2018-02-22:
OK, that sounds like a good idea. But my suggestion is not to refactor the data structures or storage. It is to refactor the code that interact with the data structures. By moving to a more object-oriented model, functions are declared as member functions, implemented in one place only, and then re-used everywhere else. This would also make the work with refactoring storage easier, as you only need to update the struct/class' own member functions/methods.

Jesper

closed 2018-06-05 by mpaperno