opentx/opentx #5712: USB

Ibiscanes, 2018-02-22:
I have upgraded my Taranis to the latest firmwire. I turn on the transmitter by holding in the trim tabs. It starts up fine, I plug in the USB cable. I turn on Companion 2.2 and try to download my models. I get and error message I  need to upgrade to 2.0 for the USB cable to work

3djc, 2018-02-22:
Could you include a screenshot of the message, because I do not understand what you are refering to.

Ibiscanes, 2018-02-22:
My Taranis  X9D 

Version 2.2.1(CCA71801)

EEPR 218

FW opentx-X9D

Ibiscanes, 2018-02-22:
I also receive this message when I open 2.2 Companion

 

Firmware opentx-x9d+ does not seem to have been downloaded Release 2.2.1(2018-01-09) is available. Do you want to download. 

I responded No. It then asked Do I want to ignore this release I responded No

kilrah, 2018-02-22:
Attachments don't work when you answer via email.

Ibiscanes, 2018-02-23:
Pictures attached to issue

kilrah, 2018-02-23:
Reread the main message a bit more correctly, not jsut the fine print...

Anyway you probably have a bad or charge-only USB cable, try others.

Also this place is for bug tracking , not support. Please use forums for support.

closed 2018-02-23 by kilrah

Ibiscanes, 2018-02-23:
I should have left my system alone. Not worth the trouble of upgrading