opentx/opentx #5718: Companion : Inprove Telemetry Tab

pointu, 2018-02-23:
Adding the ability to re-order manually the sensors for a more useful tab when working with a lot of sensors.

mpaperno, 2018-02-23:
Probably the first thing that would need to be solved is the underlying problem that the position of the sensor in the list is important.  If you move a sensor, any LS/CF/mix/etc that relies on that sensor will be invalid (anything that accesses the sensor by index vs. name).  But there is no automatic handling of that right now, neither in Companion nor the radio (you have to fix anything manually).  I think there's already a Companion wishlist item for that somewhere in the Issues.

Also cut/copy/paste/move/delete/etc actions would then naturally go along with that.

pointu, 2018-02-24:
Ok, and talking about a "pseudo" order only at showing instant only in the tab ?

mpaperno, 2018-02-24:
I think the overall consensus is that things in Companion should appear in the same order as the radio.  Re-sorting items might be OK for a "power user" but it could quickly get confusing for everyone else.  The sensors editor can certainly be improved in many areas, but I don't think sorting should be one of them.

pointu, 2018-02-24:
ok, undestand that.

Thanks

closed 2018-02-24 by pointu