opentx/opentx #5719: Radio : Deleting only discovered sensors (not the manualy/calculated sensors)

pointu, 2018-02-23:


kilrah, 2018-02-23:
There is no way to differentiate them.

pointu, 2018-02-23:
Bonjour,

Je serais absent lundi 26 février 2018, Laissez votre message en cas d'urgence extrême je vous joindrais.

Jean-Jacques Pointel
Directeur
Phone : +33 (0)1.79.97.24.00

pointu, 2018-02-23:
And only the calculated ?

kilrah, 2018-02-23:
Not sure it makes any sense if it's only that.

pointu, 2018-02-24:
When u have 2/3 calculated sensor and dozen of discovered, it should be a start.

3djc, 2018-02-24:
Well, if you delete the discovered sensors, you calculated would likely be affected too, usulay they are based on 'discovered 'sensors anyway, no ?

pointu, 2018-02-24:
yes but when you work on the radio it would be simpler than reenter those calculated sensors, my example is because a have a lot of duplicate sensors in a plane for experimentation, and during the work it can be tedious.

3djc, 2018-02-24:
Not sure about that, given all the reference issues (sensor are almost never detected in the same order), it would be very error prone to not recreate those

pointu, 2018-02-24:
ok, understanding more about detected sensors.

kilrah, 2018-02-24:
It takes what, a minute to delete the unwanted sensors?

pointu, 2018-02-24:
it depend about quantity and on the radio u use...
Also they are not unwanted, it's when u are working with a long chain of sensors, and a lot of duplicate like 3 FLVSS, 3 FAS40, 2 RPM, GPS, Vario, PITOT, on term certainly 1 RB....add to this a few calculated...
I also understand i'm not the basic case, i just give some ideas to improve when i think it's interesting.

closed 2019-11-15 by stale[bot]