opentx/opentx #5720: Unable to load luac files (out of the Simulator)

Wesie, 2018-02-24:
Up from Version 2.2.0 it should be possible to load and run luac Files to save memory.
http://www.open-tx.org/2017/05/30/opentx-2.2.0
"Add support for generating/loading pre-complied Lua scripts (.luac) to improve memory consumption and loading times"

The radio X9E (Version 2.2.0) contains no luac file after run the schript, shout down the TX and connect the TX on the PC.
I try to copy from the script the luac ( under a different name to have a real test) out of the simulator Compangion the luac on the TX.
Following things are observerd:
* In Compangion I could NOT select a luac file
* in the TX I can select the luac file ( Test.luac) but if I press the Page Button, the script is not running.

On other hand: with which external compiler I could compile an existing lua script to a luac file, which could start an run on the opentx? (and understood, that keywords from openTX are valid? )

3djc, 2018-02-24:
Compiling script with the simu works, BUT it needs the same 32 bits size as the radio. I think the only 32 bits version we distribute are some specific linux ones.

On your radio, it will have the luac generated and available localy if an luac firmware is flashed.

I'm closing this as this is not a firmware issue and this is known to work well

closed 2018-02-24 by 3djc

mpaperno, 2018-02-24:
If you can "select a luac file" on the radio this means you don't have a corresponding .lua file.  So if the compiled version cannot be loaded (due to bitness difference) then loading will fail entirely since there is no source file to use instead.  If you did have the source file in the same folder, then it would be automatically re-compiled for you as needed.

Windows version is 32b also.

It's true that in Companion .luac files can't be selected in the file menus.  But it doesn't really matter because the radio chooses the best (or only) file type automatically regardless of which extension was originally used.

Wesie, 2018-02-24:
THX for replay; I'll will check it