opentx/opentx #5728: Companion No Option for ST7920 LCD under M128 build

axljackass, 2018-02-26:
I believe ST7920 LCD and  other LCD like ST7565R, ERC12864FSF, KS108  and the default LCD ST7565P where already supported but there is no option in Companion to select other types of LCD especially for M128 DIY build.

Also is the ST7920 uses CS1?

Can not find CS1 pin the ST7920 LCD pins.

Thanks.

kilrah, 2018-02-26:
These are only selectable for the MEGA2560-based DIY radio.

axljackass, 2018-02-26:
Thanks!

Can we request to include to the next release of Companion the option to select for other LCD types for M128 build, as this LCD driver were already ported we just maximize the usage of Companion for m128.

Thanks again.

kilrah, 2018-02-26:
It's not that simple, it's not just a checkbox that needs added in compaion but actual support for the options would need to be added to the firmware for M128.

Given we're ending support for AVR boards it's unlikely any further effort will be made on them by the current team.

closed 2019-11-15 by stale[bot]