opentx/opentx #5730: stickrev - mega2560

ALFicek84, 2018-02-26:
Hi,

can stickrev option be enabled for mega2560 board? Im rebuilding old Futaba FX-18 and both my vertical sticks are reversed. No option in companion. Can´t set it. I can select/set it when i run simulation in companion, but not on real HW.

thanx

kilrah, 2018-02-26:
Given you have to do custom wiring why don't you jsut swap the pot wires?

ALFicek84, 2018-02-26:
As its not "clean" solution. INV STICK feature is already implemented for other boards. 
https://github.com/opentx/opentx/issues/660
Maybe according to fact it is working in simulator it should be working at live mega2560 board too??

Swapping pots is my last solution, as pots V+H are soldered to small PCB with common GND and VCC, which is clean in my oppinion. Resoldering will just add wire garbagge.

kilrah, 2018-02-26:
As we will be removing AVR support from 2.2 no more changes will happen there. 
We will try to make 2.1 fully functional for AVR boards and will consider enabling stickrev there.

closed 2019-08-19 by kilrah