opentx/opentx #5731: Problem Analyse different output in otx backup's. same 2.2.1

flaviomartins8, 2018-02-26:
Hi, 
Problem Analyse different output in otx backup's. same 2.2.1

Today I try to understand the airplane crash, I fly last time and  the data  that in one of the backups, the channel 2, elevator, change between INV and NORMAL.
26 Jan, 3 fev , 13 fev, 13 fev 2, 17 fev 20 fev 26 fev  ( INVERT) ( right as phoenix S8r)
19 fev, 22 fev and 22 fev 5 ( Normal)

Why different dates the channel go to invert and normal?
I use 2 computers, both MAC. 
Is there any chances, they change channels?
I reinstall the firmware in radio 2 or 3 times, last weeks and  always with  PERFIL Lua, Lua C and Multimodule.
I make one OTX with the models compare.
My radio is the Horus 10Xs.

[prova.otx.zip](https://github.com/opentx/opentx/files/1760593/prova.otx.zip)

3djc, 2018-02-27:
The most likely is that one of your Mac had a version inverted, so depending of witch you where sending from,, you got invert of not. Anyhow, I cannot see here any evidence here that this is caused by OpenTX software

kilrah, 2018-02-27:
You added the flaps and inverted that inbetween, guess is you just clicked in the elevator's invert field inadvertently.

closed 2018-03-02 by 3djc