opentx/opentx #5732: QX7S Wireless Trainer

kriskirkland, 2018-02-27:
Slave/BT mode not available to enable wireless trainer in 2.2.1 on QX7S

3djc, 2018-02-27:
Then you haven't set your BT to trainer in radio hardware

closed 2018-02-27 by 3djc

kriskirkland, 2018-02-27:
I'm saying the option is not there to enable it...unless I'm over looking it.

3djc, 2018-02-27:
Which one ? Slave/BT or trainer in radio hardware ?

kriskirkland, 2018-02-27:
I have neither option in both radio itself and companion.

3djc, 2018-02-27:
What is the exact 2.2.1 version displayed on the radio (yes, companion does not know about BT)

3djc, 2018-02-27:
[image]

[image]