opentx/opentx #5733: Horus X12S - RC mode problem

Edk8, 2018-02-28:
1. I was using Futaba mode 1 for many years and flashed my X12S with 2.2.1 and found the channel on the OPENTX mode 1 is different. It becomes mode 3 in OPENTX.

2. Set the default channel order and the mode option as AETR (mode 1 from Futaba or mode 3 in OPENTX), when I change model (plane) to another model (quad), the control stick response change to RETA however mode number and default channel are still AETR on the display. Have to change back and forth to get it right again.

If I change back to plane model, have to do the same again.

In other words, the RC mode changed when switching from plane to multirotor or vice versa.

3djc, 2018-02-28:
Channel order and mode have nothing in common. Mode defines which sticks drive (THR/AIL/ELE.RUD). That's a radio wide settings.

Channel order is per model, and defines chich CH outputs what. You can choose a default value that will be used for each model creeation, but it is and stats stored per model. If you like it, you can output throttle on ch7 on a model if you want, still your throtle stick will be defined by mode.

closed 2018-02-28 by 3djc

Edk8, 2018-02-28:
Hello!

What I’m trying to say was, the radio should has a consistent mode (say mode 1 with AETR from ch1-4) throughout all model no matter I change to plane, Heli or quads. Right?

I have 3 models in “Quad” category and 2 models in plane.
Now my problem is when I have the correct mode (AETR) in Quad, when I select one of the plane model the radio become RETA. 
If I change it in this plane model to AETR, when I select the Quad model, the radio will have RETA again.
It is always opposite. when you change one side (Quad) to AETR then the other side (say plane model) will becomes RETA and vice versa.

So every time I have check the mode when I change model.

Regards,
Edmond Kwok

s700ky, 2018-02-28:
> Are you running OpenTX ? There is no such things as predefined categoris in OpenTX, there is no quad or heli or plane mode.
>
> Each model has its own channel order, wich is copied at creation from the radio defaults you can setup.

Mine has categories?

3djc, 2018-02-28:
Yes, but no predefined or mode. You can create glider, quads, car, boat or whatever, they do not have any impact on model

s700ky, 2018-02-28:
No. They are just storages areas to keep like models together so they are easier to access.

tomkfly, 2018-02-28:
You can select whatever default channel order you want. However, models already setup retain the original settings.