opentx/opentx #5734: R9M has no bind button in current build

x4FF3, 2018-02-28:
Hi, 

yesterday i tried to bind R9 with 2.2.1(build from 23.02.18) on x9d. but no bind button was shown. i reflashed the build from 18.02.18 and the bind button was shown again.

KR
David

3djc, 2018-02-28:
Please include photo, because it seems ok here, and there as been absolutely not change in the area between the two

3djc, 2018-02-28:
Simu does not show an issue either
[image]

kilrah, 2018-03-01:
Not only *in the area", not a single line of code was changed in the firmware between the 14th and the 24th.

x4FF3, 2018-03-01:
18-02-18
[image]
[image]

18-02-23
[image]
[image]

3djc, 2018-03-02:
Those are not our compilations, right ? You can see from the code stamp (cca718008) that you are using the same source code, if it does not produce the same result, it means soemthing is wrong with your compilation

x4FF3, 2018-03-02:
both versions were downloaded via companionTX. 
options of the version with bind button: opentx-x9d-ppmus-multimodule-noheli-lua-luac-en
options of the version without bind button: opentx-x9d-multimodule-noheli-lua-luac-internalppm-en

kilrah, 2018-03-02:
Aha, it helps when we actually know what you're doing. 

Then yes, external module lines seem to be broken when internalppm option is selected. 
Since there likely is no reason at all for enabling it in the first place you can use firmware without it in the meantime.

3djc, 2018-03-02:
As a reminder, internalppm is for people who have removed the x9 internal module to remplace it with something else

x4FF3, 2018-03-02:
glad i could point out a bug.

3djc, 2018-03-02:
Yes, thx, sorry, I was thinking nightly based on your report title, glad we sorted that out. Will be fixed shortly

PR #5742 (2018-03-04): Fix R9M missing bind when internal ppm is selected
Also removes that choice from x7 in companion since is was never coded for 128x64

This fixes #5734
closed 2018-03-04 by kilrah