opentx/opentx #5735: X10S sync overwrites model files

ufoDziner, 2018-03-01:
Hello,
  I normally make model changes in the field on the radio, and in companion when at home.  If I make changes at the field, I import the models, then save them to the local .otx file.  If I make them in the companion, I save to the .otx file, then write the models and settings to the radio.  Both of these work well.  The problem comes with sync.  It seems that no matter which way I have the SD card sync, the model files on the radio are overwritten by old files.  So, after a sync, I have to write the models and settings to the radio again.  If I want to have the current files that I just wrote to the radio synced with my local SD mirror, I have to copy them manually.

It seems like there needs to be a way to separate/specify which set of model files to use in the sync vs. the rest of the files.  This way, it would be easy to use the model.bin files off of the radio to sync with the local SD, but send the new local SD files to the radio.

Thanks

ufoDziner, 2018-04-12:
@raphaelcoeffic I'm attaching some screenshots to show the settings and results.  The otx1.png shows the settings and and my saved .otx file.  Just before taking that screenshot, I imported the model settings from the radio and saved them to my .otx file to ensure my local file is the most current.  I then imported them again (which shows as "document2.otx" in the third image).  Next Icompleted a sync using the sittings in the first image.  The second image shows the summary after I completed the sync.  Finally I imported the model settings from the radio after the sync ("document3.otx" in the last image).  As you can see, my radio settings are now different than what I had before.  I had made no recent changes locally, only on the radio.  It seems that no matter what I do on the radio, it will always get overwritten with the local settings during a sync.  

[image]
[image]
[image]

raphaelcoeffic, 2018-04-12:
@ufoDziner there must be an issue with the timestamps. Could you please check the timestamps for the files on the SD card and in your local directory after each of the steps you took? At some place, something must be screwed there.

mpaperno, 2018-04-12:
For 2-way sync to work, the timestamps must be correct/match on both file systems.  The fact that it has something to sync right after running a sync suggests they're not.  (Unless the prior run was also in test mode, of course.)

It might be useful to add a file/folder name filter to the sync settings.  Can't remember if I only thought about it at the time, or if I left a hook in place already (just no UI).

ufoDziner, 2018-04-14:
Okay.  I did another test today.  **Beware of all the screenshots.**  The sequence of events is as follows:

Deleted a unwanted model from the radio.
Start opentx.
Open my current .otx file.
Connect radio.
Import model settings.
[image]

(files before any sync)
[image]

Save new imported model settings as current.
[image]

Copy a new sound file to the Radio's SD card (Drive D:).
Sync files with radio as the master.
[image]

[image]

Make some changes to the models in companion to use the new sound file I copied to the SD card.
Save the .otx file.
Write models and settings to radio using the appropriately named button.
[image]

Sync again to get my local SD backup files updated with the model settings that I just wrote to the radio.
[image]

[image]

As you can see here, it reverted some of the files back to some other form.
[image]

I then had to use the write models and settings button again to get the files on the radio correct.
[image]

So, I'm not sure what the sync is doing.  As I understand, when I make a change to my .otx file, and write that to the radio, it changes the model files on the radio.  Nothing changes the local SD card backup except for sync.  Which means if I make a change in companion I need to accomplish the following steps:

Make change in companion.
Save updated .otx file.
Write model and settings to radio.
Sync SD card with radio as the master to update the local files to match my new radio files.

Please let me know if what I'm doing is wrong, but I'm pretty sure something is broken.

Thanks

kilrah, 2018-04-14:
Can yu post your 3 companion settings tabs?

mpaperno, 2018-04-14:
Why are some of the files timestamped 2019?  Either the computer or radio seem to be living in the future... ;-)

raphaelcoeffic, 2018-04-14:
@ufoDziner you should basically delete your local SD backup, and re-sync. If your PC and radio are using the same time, it should then work. The old files on your PC with timestamps in 2019 are the issue: they will get never get overwritten, and the issue will first be solved in 2019 ;-) The question is: how did they land on your PC?

@mpaperno probably not directly related to this issue, but wouldn't it be a good idea to be able to sync the time from PC with companion? Or at least detecting a timestamp issue before synchronising?

mpaperno, 2018-04-14:
@raphaelcoeffic Uh, sure, that would be nice...  :smiley:   We'd need some kinda API for that!  (Or do we have that somehow?  Besides the "devs-only" CLI.)  When we're syncing with SD card the radio is most likely off anyway.  But... yet another option to add to sync would be a maximum time difference.  (And make it ignore system files somehow!)

raphaelcoeffic, 2018-04-14:
@mpaperno I just opened #5829 for that purpose.

raphaelcoeffic, 2018-04-14:
@mpaperno ignoring sys files is a good idea anyway ;-)

raphaelcoeffic, 2018-04-14:
@mpaperno by the way: what about correcting time stamps situated in the future while scanning / synth the SD card (with some acceptable delta).

ufoDziner, 2018-04-15:
@raphaelcoeffic Okay.  I created a blank local SD directory and told companion to sync from the radio to the local folder only.  Here are the results.

[image]

[image]

[image]

The 2019 files are coming from the radio.  Do I need to delete them there first?

raphaelcoeffic, 2018-04-22:
@ufoDziner you probably don't want to kill these models by deleting the files. But you could try to localise to which model they are tied, make some modifications to these models so the date gets written properly, and then sync to blank directory on your PC. Please control that all files on SD have a correct timestamp (NOT in the future), before doing the sync to your PC.

closed 2018-04-24 by raphaelcoeffic