opentx/opentx #5741: incorrect GPS display

hobby1946, 2018-03-02:
Transmitter Horus X10 OTX V2.2.1
The GPS data is only partially displayed.
The N position is always 0!
In telemetry and in the display
This is sometimes the case with the original FRSKY GPS

kilrah, 2018-03-02:
Displayed where, using what widget/page,...

3djc, 2018-03-02:
We need exemples, like simu or photo. It might depend from your position and/or setting (nmea, dms)

hobby1946, 2018-03-05:
It might depend from your position and/or setting (nmea, dms):
 the error is the same for both settings

3djc, 2018-03-05:
From your issue title, I thought you had a display issue, but from what you present here, you gps data is not read correctly. Could you try logging it and see if value are 0 too ?

hobby1946, 2018-03-05:
ll try that.
Where can you check that?

The display works with X9D + and V2.1.8
With X9E and V2.1.9 no more.

hobby1946, 2018-03-05:
The value GPS1 is not logged although ticked
The field GPS1 can not be found in the LOG

hobby1946, 2018-03-06:
[image]

[image]

3djc, 2018-03-06:
All the info here make me think there is an issue with the telemetry sent to OpenTx, not in OpenTx itself.
On X10, in statistics/debug screen, the last field is Tlm RX Errs, what is the value ?

hobby1946, 2018-03-06:
Sorry, where is the debug screen?
What do I have to dial?

3djc, 2018-03-06:
Long enter, statitstics, pgdn

hobby1946, 2018-03-06:
OK.
Now while trying (but not always)

Display after switching on transmitter and receiver

[image]

[image]

Display after reset flight data

[image]

[image]

kilrah, 2018-03-06:
Can you say which GPS sensors they are and where they're connected?

hobby1946, 2018-03-06:
GPS1
von Naza Lite über Zaggometry  (0x800/15)

GPS2
Frsky GPS (0x800/4)

kilrah, 2018-03-06:
And what happens if you connect only one of them at a time?

hobby1946, 2018-03-06:
Nothing changes when I connect the GPS individually.

GPS1 almost always works

GPS2 almost never (not even at 0x800 / 4)
but all other values are correct
only GPS-N is 0

kilrah, 2018-03-06:
and if you create a new model and only connect/discover the frsky GPS?

hobby1946, 2018-03-06:
the Frsky GPS is displayed to 99% correct.
this is also the case with a new model with only GPS

the other GPS shows 99% no N position
even with a new model with only GPS
All other data is ok

The same model (copter) attached to a X9D + with V2.1.8
displays both GPS position correctly

hobby1946, 2018-03-06:
ich habe den Copter an eine X9E gebunden.

bei otx V2.1.8/9
hier erscheint keinerlei GPS1 Wert (nur GPS1)

[image]

[image]

bei otx V2.2.1 erscheint nur die N-Pos nicht

[image]

hobby1946, 2018-03-07:
Hello

I got a X9D + and the same copter was bound.
with OTX V2.2.xxx also the N-position was missing

But with OTX V2.1.8 all GPS data was available!
Also the N position!

[image]

[image]

The Zaggometry sends all data properly

hobby1946, 2018-03-09:
here the log file of my X10 with v2.2.1
only the E position is displayed
the N position is 000 ..

[image]

hobby1946, 2018-03-11:
Here is a data protocol from the S-Bus at the receiver
GPS ID 0x800 - SubID 15
The transmitter does not indicate an N position

[GPS1-0x8E-15 Zaggo.zip](https://github.com/opentx/opentx/files/1800776/GPS1-0x8E-15.Zaggo.zip)

hobby1946, 2018-03-15:
Hello
the error is localized.
In the telemetry sensor falsely no byte stuffing is performed!

The transmitter determines for the north position (in my place) an incorrect checksum and therefore North is then displayed with 0.

closed 2018-03-15 by hobby1946