opentx/opentx #5743: Make failsafe available on SBus

pindeslandes, 2018-03-03:
Taranis X9D+ - OpenTx 2.2.1.

When trainer signal is lost, a flag in the SBus protocol is set and OpenTx uses it to disable all TRx inputs, deciding that there is nothing good to do with failsafe.
But the user may want to decide by himself what to do with failsafe.

geofrancis, 2018-03-03:
What do you mean by decide what happens with failsafe? If it looses trainer connection it switches back to master handset. What else can it do?

pindeslandes, 2018-03-03:
For instance, when the link between the student and the teacher is broken, it’s most often because the two transmitters are too close. Then I just want to have a voice alert and to hold the TRx values, letting the teacher decide whether He has to take control (if the plane is in a critical situation) or to wait a couple of seconds the link to be recovered if there is no danger. That avoids disturbing the student for nothing that couldn’t be dealt by the teacher.

And if I used a CPPM receiver I would get the failsafe so why should it be forbidden if I used a SBus one?

closed 2019-11-15 by stale[bot]