opentx/opentx #5744: Taranis X9D+, External XJT, OpenTX 2.2.1, no option with telemetry.

Nihonski, 2018-03-03:
Hi,
I am using Taranis X9D+ for long time with external XJT modules and telemetry enabled.
Testing OpenTX 2.2.1 recently, I cannot bind receivers with telemetry enabled any more.
Please make all options available, same as for internal XJT.
Generally I welcome the binding options, very convient compared to receiver jumpers.
Thank you for the nice feature.

mpjf01, 2018-03-03:
You have to turn off the internal iXJT now if you want telemetry on the external module. Or not use 2.2.1.

Nihonski, 2018-03-04:
I am using both, internal and external XJT with seperate telemetry.
Internal XJT telemetry is displayed on the Taranis, the external XJT is displayed on a Dashboard.
The S.Port line from the external XJT to the Taranis is cut, so no interference between both telemetries.
Moreover the internal XJT and external XJT are synchronized by additional electronic, so transmitting and receiving occurs at exactly the same time to reduce sporadic interference of adjacent RF channels between both RF-links.  
My system based on X9D+ provides true XJT and receiver redundancy, with two telemetries allowing the two RF-links to monitor each other.
With your new restriction, my system becomes obolete with OpenTX 2.2.x. , so sad.
I really hope you could consider my issue and keep OpenTX open for everybody, please!

I hope to switch from 2.1.9 to 2.2.1, as recent sensors like the RB10 are not handeled properly by 2.1.9.

3djc, 2018-03-05:
Then you can understand you are in a very,, very specific situation., You have a fairly simple workaround, you turn off internal XJT when you are binding the external receiver

Nihonski, 2018-03-05:
Assuming that there would be no "side-effects" in OpenTX, your suggested workaround is completely fine with me. Thank you very much for your support.

3djc, 2018-03-05:
I'll be closing that then. I do not anticipate the workaround to give you any trouble, but if it does, please let us know !

closed 2018-03-05 by 3djc