opentx/opentx #5745: Add a volume parameter to lua playTone() function

pindeslandes, 2018-03-03:
Taranis X9D+ - OpenTx 2.2.1.

You use this function when you have some specific need so you certainly need to set its volume too.
For instance, I have designed my own vario sound in a script using playTone but I find it too loud compared to the other sounds: I would like to lower it (the same way as for the original vario sound).

kilrah, 2018-03-03:
These will play using the "beeps" volume. 
It is not really possible to give lua control of that without some potentially undesirable implications.

pindeslandes, 2018-03-04:
Good to know, thank you. And good news for me as I don’t use beeps!

kilrah, 2018-03-04:
Well you use them a bit without realizing (button presses, trim key beeps etc)

closed 2018-03-04 by kilrah