opentx/opentx #5747: Confirm antenna is connected before powering external module.

geofrancis, 2018-03-04:
Pretty much any uhf system will be destroyed if the handset powers on without an antenna connected. So if there was a warning that makes you confirm the antenna is connected before it powers on the external module. Would stop a lot of modules being damaged.

Ideally it should be a configurable option since 2.4 modules generally have short antennas and can be left in place.

Graupner radios do something similar where they won't transmit untill you hit ok on the screen.

3djc, 2018-03-05:
Maybe it does not require yet another model setting. We are considering reviewing model powerup / change sequence so that model notes are displayed before turning on module. That would be a simple way to achieve this (and much more) without making the UI more complex

3djc, 2018-03-05:
Not quite, because meny will complain that their external module does not require such a check (like multimodule for example)

geofrancis, 2018-03-05:
Yes that's why I said that it would have to be an optional check. Since 2.4 modules usually keep the antenna on all the time. But uhf do not.

geofrancis, 2018-03-05:
In not really bothered how it is done but it's something that needs to be done especially now tbs and frsky are pushing modules to racers who generally stuff equipment into small bags that require removing antennas.

geo502, 2018-05-04:
I agree with @geofrancis ,  A lot of the Crossfire users haven't completely changed over all of their models to use crossfire so they have a mixed bag of models using internal and external TX modules.  I keep my transmitter in a bag and have to take of the external antenna when I pack away and  have flown away 1 too many times now with no antenna on and ended up with failsafe 50 yards and crashing to the ground. I am looking into options to help myself remember this now, like a preflight-checklist or safety switch with custom functions but  an actual warning that is optional like the switch or throttle warnings during startup would be a very nice feature.

PR #5758 (2018-06-09): Disable pulse while model notes are displayed
This keep pulses off while model notes are displayed. This allow users to writes things like "Don't forget to connect external antenna', or 'this model needs module xxx' before those module are turned on.

This works at  boot and also when changing model.

Unlike other warning, getting out of this one requires exit/rtn

This closes #5747
closed 2018-06-09 by bsongis