opentx/opentx #5748: Multi: FLYSKY AFHDS2A RSSI value issue

pascallanger, 2018-03-04:
Hi,

I've got multiple reports that the RSSI is wrongly reported when using the protocol AFHDS2A.
The value is extremely low which makes the radio alerting at really short range for nothing.
Can someone check it?

Background: Multi is sending the raw telemetry data received without touching it to OpenTX. The parser is within OpenTX.
You could read the RSSI value like I'm doing here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/fa9e559195b4e97df7a43a8ae84b4014774b4e7e/Multiprotocol/AFHDS2A_a7105.ino#L101

Thanks, Pascal

MikeBland, 2018-03-04:
For information, on ersky9x, I take the raw RSSI value and convert it to 135-raw_value.

Mike

pl4hh, 2018-03-04:
just my 2 cents since i'm currently reading up and setting up one of my old flysky equipped quads. 
i can at least confirm that rssi seems really low, have a fs-x6b rx bound in fs afdhs2 ppmibus mode.
telemetry is received but get alot of low rssi or telemetry lost warning from 0-100meters, haven't really flown a range test yet. need to figure out the failsafe first. which does not seem to work. rx still sends signals when tx is off. also using a january ordered IRX 4in1 plus, if that might be of help.

paultbg, 2018-03-04:
I'm using the latest Opentx on my Taranis x7 and while the Multimodule works well my IA6B receivers the telemetry is showing really low (one single bar on the main screen). In range check mode the RSSI is starting from 47 (TX one meter away from the RX) and if I cover the antenna with my hand it will go to critical, Rssi around 30.

schwabe, 2018-03-04:
Flysky has quality which has a diferent range than RSSI from Frsky
I don't think that faking a Frsky like RSSI is the right way to do. The right thing is to have the right rssi warn/critical values, but I don't really know what they are for flysky

kilrah, 2018-03-04:
IMO that is for users to adjust the 2 levels appropriately.

MikeBland, 2018-03-04:
The RSSI value in AFHDS2A is reversed in magnitude, a small value is good, a large value is bad.
This needs to be mapped in such a way that the 2 levels can be set. From running companion, it seems they cannot be set below 20. For the way the value is currently mapped, I think these are too high to allow a correct adjustment.
In passing, DSM telemetry uses a Tx RSSI value from 0 to 31, so, unless this is mapped to some other values, 20 is again too high.

schwabe, 2018-03-04:
Yeah, but Flysky the RSSI value isn't the interesting one, right? Original Flysky use the Quality value as indicator of the link quality. So I think OpenTX should also use quality instead of RSSI and that is the right direction.

pascallanger, 2018-03-05:
If you implement LQ then some other protocols would benefit from it. I'm thinking at FrSky TX side (D has it and it could be added to X) and Bayang both RX and TX.

schwabe, 2018-03-05:
@pascallanger I think you are thinking about a different link quality value. FlySky has Linkquality as dedicated telemetry value that comes from the RX.

pascallanger, 2018-03-05:
@schwabe This is the same for Bayang. The RX is sending by telemetry to the TX (and the TX does the same on telemetry recieved packets) a link quality value which is based on a counter incrementing every packet received (every 5ms) over a 1 sec period.
Today I'm feeding the RSSI value with it but this is a link quality value.
FrSky does not implement it on the RX side but we have it on the TX side.

wannaduino, 2018-03-14:
And??is there now an solution for us? i see this is now closed? i just seen this NEW tread in RC groups, but yet closed again?

3djc, 2018-03-14:
What was said in this thread is that we do not intend to somehow manipulate the RSSI data to make it look in similar range as FrSky. You can adjust  the range to match what a particular receiver is giving. Since you did not seem to have contributed to this thread, I cannot be more specific for your particular issue if any

wannaduino, 2018-03-14:
Thanx, you are 100% right. about the last part. ( my bad ) this is new to me, but i will be more carefully when responding.
Yes, i have the Multi 4 in 1 the Blue one with the Atmega, and Yes i have the Taranis QX7 with the obnoxious blabbing of the RSSI because of the value 60 or so. (receivers,  flysky X6B 6ch,-Tellemetry)
But i read it can be set higher to 135 or so, so my question would be a link to read how to do, or tell me where to change on the radio.
WannaDuino

3djc, 2018-03-14:
I'm really sorry, but I not sure I fully understand. Why would the radio complain at 60, at it is a very common value, including for FrSky ? (usual warning occur with rssi lower then 45/42, and that limit can set anywhere from 15/12 to 75/72)

wannaduino, 2018-03-14:
That is why this tread is opend my friend,  or do i miss something LOST IN TRANSLATION OR SO???
Pascal know whats up, i am here because of him opening this up VIA RC GROUPS.

schwabe, 2018-03-14:
@wannaduino Get me the default values for quality from a FlySky radio for warning and critical and I can set them as default in OpenTX for flysky too. Otherwise you will have to set/guess them yourselves

kilrah, 2018-03-15:
Thing is we can't adjust range far enough currently, not enough bits in struct...
And @wannaduino it wasn't closed, look correctly.

lephiloux, 2018-11-12:
Hi, i have the same issue if i look at code now the RSNR is used as RSSI but don't think it's the best value to take with actual configuration range. I attached one of my flight logs to give you more information about range change.
[lisam-opentx-logs.zip](https://github.com/opentx/opentx/files/2573363/lisam-opentx-logs.zip)

shurik-filippov, 2019-01-31:
Yes, there is a problem, tested on 3 receivers x6v. Because the value is too low and it would be more acceptable to use the value of RNse.

PonyOny, 2019-06-18:
I'm having the same issue with "iRx4 Lite" bound to a FlySky X6B, I get all the time alerts because of low RSNR values, as @shurik-filippov mentions the RNse value seems more reasonable.

fremebondo, 2019-07-05:
We can, of course, set lower alarm threshold for RSSI on OpenTx, but the signal bar in the main page always shows only 1 or 2 bars. I found no way to modify this. Should this issue be referenced inOpenTx github issues?

fremebondo, 2019-07-05:
> What was said in this thread is that we do not intend to somehow manipulate the RSSI data to make it look in similar range as FrSky. You can adjust the range to match what a particular receiver is [...]

We can, of course, set lower alarm threshold for RSSI on OpenTx, but the signal bar in the main page always shows only 1 or 2 bars. I found no way to modify this. May the bar scaling be related to de set alarm level instead of using absolute "bar size"?

popado, 2019-07-18:
FlySky FS-iA6B Receiver 6CH with Jumper T12 (Jumper 4in1), RSSI 39-40, low signal or lost signal alarm all time, after binding. Servo seem to work fine. Thanks.

DarkSoulRL, 2019-09-17:
Fli10 reciver with Jumper multi module. AFHDS2A protocol. Low signal or lost signal alarm all time, after binding.

pascallanger, 2019-09-18:
I have a fix for the low and critical alarms. I'm working on adding a lot of sensors. It's currently under testing before a PR to OpenTX.
To be able to access the new sensors you also need to upgrade the Multi module to 1.2.1.77 or more.
Pascal

DarkSoulRL, 2019-09-21:
> I have a fix for the low and critical alarms. I'm working on adding a lot of sensors. It's currently under testing before a PR to OpenTX.
> To be able to access the new sensors you also need to upgrade the Multi module to 1.2.1.77 or more.
> Pascal
Can I get firmware 1.2.1.77 for tests? I have a Jumper T16 multimodule

pascallanger, 2019-09-21:
The github repo is up to date so you can build the version 1.2.1.79 by yourself. If you need a binary I'm going to do a new Multi release some time today.
OpenTX PR for additional sensors and fixes is still not submitted but you can build it from my branch: https://github.com/pascallanger/opentx/commits/sweet-pascal
Pascal

3djc, 2019-09-21:
I would very strongly advise against flying with that branch, it is not even ready for nightly

nekoth, 2019-11-20:
Was there any progress since the last update?
I have x6b receiver and RSSI is almost always -60Db. It sure gets lower if I go much further from drone, but it is -60Db almost everywhere near.
I was able to add RSNR input with 45dB scale and send it as a channel back to receiver (weight 200%, offset -100%) to show it through BF OSD. Kinda makeshift RSSI. But it very imprecise this way, compared to my old FlySky i6x with custom firmware which was sending a rather precise RSSI as Err% channel on CH11.
I have updated my internal 4in1 to 1.2.1.85, but to no avail. Using official JumperTX 2.3.

pascallanger, 2019-11-20:
You are using JumperTX and posting on the OpenTX github...
Anyway, you need to update to OpenTX nightlies to get all the Flysky AFHDS2A features.
And by the way you also need to upgrade your multi module to the latest 1.3.0.44.
Multi can send LQi directly to the RX if you compile the code by yourself and enable #define AFHDS2A_LQI_CH 14 in the _config.h file.
Pascal

3djc, 2019-11-20:
Closing since this got fixed in nightlies

closed 2019-11-20 by 3djc