opentx/opentx #5750: Companion compare models enhancement suggestion

timrowledge, 2018-03-05:
The 'compare models' feature is very useful but (at least on my iMac and for my eyesight) somewhat hindered by the too-subtle visual differences in use. Given the older demographic of model flyers I doubt my sight is drastically worse than the average!

Whilst the soft grey is nice for the 'plain' text, it blends too nearly into the green used for the changes in the left-side model details. The red is a bit easier to see when there is a substantial amount of it. The combination makes it quite  strain to work out the differences.

I'd like to urge some further cueing for the changes. A simple to implement option that might be enough would be adding underline to the text style. A 'change bar' at the outer edge of the two columns (as used in many word processors) would be another helpful cue. A more complex solution might be the option to only show the entires with changes; that really makes it clear!

A much more sophisticated approach - with obviously more work involved - would be a scheme typiifed by the text diif tool in TextWrangler. See https://i0.wp.com/subharanjan.com/wp-content/uploads/2014/09/Compared-Differences-TextWrangler.png for an illustration should you not be familiar with this. The ability to move versions from one side to the other (see the Apply buttons in the image) would be interesting too.

mpaperno, 2018-04-07:
I agree about the colors, though I don't like the dim gray default color either -- not easy to read.  The color coding is also probably not great for "color challenged" people.   Font styling could help for sure (underline/bold/italics/size).  I also like the idea of change markers in the gutter, though I'm not sure how easy that would be to implement.

"Show changes only" is a great idea.

Integrating a diff tool would probably be a bit much (we'd need one that's cross-platform already), but perhaps one could save the printed versions to files and then use an external diff tool to compare.  I guess they'd need to be plain text, or use a diff tool which understands HTML or PDF.  Editing models using a diff view is probably not going to happen anytime soon.

elecpower, 2018-04-07:
In the next 2.2 release the gray is replaced by black (tick).
Colors and font styling could be user selectable and saved in preferences.
Highlighting changed lines in gutter would be a coding challenge for not a lot of benefit if the colors and styling is implemented.
Integrating a cross-platform diff tool is unlikely to occur when in the current version individual model prints can be saved to html and therefore can be loaded into diff tool of your choice.

timrowledge, 2018-04-10:
The good news is that there is already a 'cross-platform diff tool' in Companion in order to do the highlighting and that's probably enough for now. Cleverer would be nice but not essential.

Whilst user preferences for fonts and colours would definitely be nice I'd urge starting with just stronger colours and bold/underline to start with. Sometihng simple soon and better later is always preferable to nothing for ages and fabulous much later.

I'd even take a look at the code and offer changes if my tolerance for C code hadn't been eroded away so much. Maybe if I have some idle time.

elecpower, 2018-04-10:
The colors and style are affected by the OS, themes, screen resolution, etc and then you add in personal preferences and visual challenges. If there is a move to change the formatting further then there is not a large amount of extra effort to include user preferences.

mpaperno, 2018-04-10:
Might be worth looking into using an external style sheet (CSS) to manage the formatting.  I think the CSS support in QTextDocument is limited, but it may be enough for our basic needs.  Perhaps we could provide a couple/few predefined styles (selectable somewhere in Preferences or even right in the print window), and for ultimate flexibility a user could customize their own.

mpaperno, 2018-04-10:
I'm generally a big fan of putting formatting instructions into separate resources, when practical.  And it's not uncommon for apps to have user-editable "themes."  But even if they're not directly user-editable, it's still a nice separation.  

(I did something similar for the translation files, where you can configure a env. variable with a path to local files and CPN will look there first, so one doesn't have to rebuilds CPN to test a translation.)

I think the level of CSS we're talking about here would be pretty minimal (and probably easier than editing an OTX Model!  :) ).  Some font attributes basically.   Assuming QTextDocument even supports CSS classes.

Looks like it does! 

>All CSS 2.1 selector classes are supported except pseudo-class selectors such as `:first-child`, `:visited` and `:hover`.

http://doc.qt.io/qt-5/richtext-html-subset.html  (CSS support at bottom.)

And it specifically supports adding external "StyleSheet" resource types.  http://doc.qt.io/qt-5/qtextdocument.html#ResourceType-enum

elecpower, 2018-04-11:
Thanks for the research. I'll do a bit of experimentation with the current code and external css to see what is practical and works with relatively minimal effort.

kilrah, 2018-04-11:
> Sometihng simple soon and better later is always preferable to nothing for ages and fabulous much later.

Our project however usually follows the 2nd philosophy, better do it good once even if it means waiting until that good solution is found instead of faffing around multiple times until you get to a decent point, with the code maintainability/evolutivity issues and user confusion that this brings.

Another similar project uses the "quick and incremental" approach and has cornered itself into ugly things several times because of that.

PR #5866 (2018-05-21): Model print and compare user customisable style
New feature to allow users the ability to customise the print style via a stylesheet. The main use is for model compare to more clearly highlight differences.
A reuseable simple file editor has been included to manage the customisable file.
The default is a resource file to ensure it is kept in sync with the code and any QT changes. The initial stylesheet version replicates the code.
The customised file is stored in the OS application data folder so it persists across Companion versions.
QT 5 has a number of limitations when using stylesheets.

F [... cut]
closed 2018-05-21 by bsongis