opentx/opentx #5751: Taranis X9D+: Problem with crossfire.lua: forever Waiting for Crossfire devices

flexd, 2018-03-05:
Hi there. I'm not sure this is a code issue yet, but I'm posting here for lack of options.

I am on OpenTX 2.2.1 (cca71808), with the sdcard contents for that version.

I have a TBS Crossfire Micro module with the latest firmware (2.25 Beta), and I have also tried using the module with 2.12 and 2.24 (latest stable).

No matter what I do, my Taranis will only tell me "Waiting for Crossfire devices" when I try to use the crossfire.lua script to configure the module.

I have not used this module to fly anything yet, but it seems to bind successfully to a micro v2 receiver.
Inside the module bay of the Taranis, '2016/12/21' is stamped, which I'm assuming is when it was manufactured.

Am I doing something wrong here?  Any ideas? As far as I can tell I have done everything the various guides about this say I should, but for them it's working and for me it is not.

Thanks for getting back to me.

flexd, 2018-03-07:
Someone suggested maybe Frsky started putting the slower inverter in the newer X9D+ radios as well? That would be annoying.

schwabe, 2018-03-08:
You can check the telemetry error statistics in the statistics to give you an idea if you have a problem

flexd, 2018-03-08:
@schwabe Tlm RX Err is 0. It would appear I have no working telemetry with the receiver though, so all signs are pointing to either the radio or the module being defective.

3djc, 2018-03-11:
Given conversations we have had over rocket.chat, it seems clear this is not an opentx issue, so closing this

closed 2018-03-11 by 3djc

walounator, 2018-03-19:
I am running into the exact same issue with my brand new module and radio. Have you found any solution? Thanks

flexd, 2018-03-19:
This turned out to (probably) be a issue with my module, or *maybe* with newer modules. I tried lots of other crossfire modules in my radio and it worked fine, and we tried my Crossfire module in other Taranis X9D and QX7, and it would not work in any of them. I only got it to work (or at least communicate with a radio) by putting the module in a TBS Tango.

Do you have any friends with crossfire modules, Taranis radios or TBS Tango radios? Ask to try their module in your radio, and so on, to see if the problem is the radio or the module.

walounator, 2018-03-19:
Thanks for your quick reply. Unfortunately I am quite alone in my region.... Now my module doesn't even light up when plugged in the radio but still does when connected to my computer. I am really screwed...

flexd, 2018-03-19:
@walounator Aha! That is a clue! Mine lights up fine. Did you remember to disable the internal module and set the external module  to CRSF in one of your models? Have you checked if the pins on your radio are bent? And are you entirely sure all the pins line up completely and hit the right holes? You could try to carefully bend the pins to make sure they have good contact inside the module. Be careful, it would be annoying to have to replace those.

walounator, 2018-03-19:
My bad, it was indeed a pin badly aligned. I prefer that compared to an RMA. Thanks for your quick support !!!

flexd, 2018-05-03:
This turned out to be a defective micro TX. New module works without having made any changes 👍