opentx/opentx #5757: Horus missing telemetry packets

rimb05, 2018-03-11:
When running a widget lua script on an x10, the telemetry seems to miss some packets.  The same script ported to an X7 works fine with no missing data.

This is especially noticeable when using Passthrough mode connected to an Ardupilot.  9000 packets (Mavlink messages) are often missing characters for example.

Packets that do come through pass the CRC in checkSportPacket(), so this isn't an issue with corruption.

The same problem happens with both internal and external RF modules (I tested both).

3djc, 2018-03-12:
Could you give us more details to try to replicate ?

rimb05, 2018-03-14:
Sorry, this was caused by an unrelated issue on my end.

closed 2018-03-14 by rimb05