opentx/opentx #5763: OPEN TX RESET ALL MY RADIO

cbsbedoya93, 2018-03-13:
ive been havinf this problem since i got the horus 10, intall opentx everything works perfect but suddenly any day i turn off the radio and when it comes back on again is a new radio. no configurations, no models, everything is gone. it happen 4 times now, and its driving me crazy. im starting to think that there is a problem with open tx or this radio is just crap. any one know what could be happening this?

kilrah, 2018-03-14:
MicroSD card probably defective, everything is stored in there. Try replacing it.

closed 2018-03-14 by kilrah

cbsbedoya93, 2018-03-14:
i tried 3 different sd cards. same thing

kilrah, 2018-03-14:
Then probably a hardware defect on the radio, see your dealer for warranty.