opentx/opentx #5764: Horus lua drawLine function doesn't work properly with diagonal lines

rimb05, 2018-03-14:
When writing a widget script on a Horus x10, the lcd.drawLine function works properly for straight lines: for example lcd.drawLine(10, 20, 10, 100, SOLID, CUSTOM_COLOR).

But if you try something like this it fails: 

lcd.drawLine(10, 20, 30, 100, SOLID, CUSTOM_COLOR)

The specific failure is that the line is drawn on the opposite side of the screen.  Upon further inspection it's as if the coordinate system gets flipped.  In other words (0,0) is bottom right when drawing diagonal lines.

3djc, 2018-03-14:
I do not have an x10, but at least on simu, it seems ok:

[image]

rimb05, 2018-03-14:
Yes, I forgot to mention that it works fine on the simulator, but not on the radio.  Made it very hard to discover!

3djc, 2018-03-14:
Ok,, thanks for the precision, we will look into that then

closed 2018-04-22 by raphaelcoeffic

Massild, 2018-05-03:
I know that this does in fact fix the issue, however it is fixing the symptom not the cause.  This fix is looking for a special case (X10 and not sim) and modifying the output.  The real issue is why is this happening in the first place.  Is there someone who is working on the PIXEL_PTR() directly to resolve this?  I am at a loss why the memory pointers would be different between the X12 and the X10.

3djc, 2018-05-04:
Because x10 screen is actualy reversed compared to x12

Massild, 2018-05-04:
Good to know, that was an unfortunate design choice by FrSky.  Do you know why this only does not work for diagonal lines?  It seems like the line function should work or not work depending on the radio type and not be dependent on if the X values are the same.

raphaelcoeffic, 2018-05-04:
@Massild digging a bit more into the code, you will see that graphics functions on X10 are always slightly different at some point compared to X12.

Massild, 2018-05-04:
Thank you @raphaelcoeffic  and @3djc  for your responses.  I appreciate them.

PR #5837 (2018-04-22): fixed PIXEL_PTR() macro on X10
fixes #5764
commit 697403b8 (2018-04-22): fixed PIXEL_PTR() macro on X10

fixes #5764