opentx/opentx #5767: horus x10s and the new opentx brick

sw3Dp, 2018-03-15:
first, i never expected opensource software to be perfect or even work, but i really loved opentx working flawlessly on my x9d for past 3 years, did and would pay for it to work again.  (got spoiled milk?)

new horus x10s flashed with opentx 2.2.1 critical error in channel output (sticks/gimbals) along with all the speaker noise and rf interference of older <2 builds on the x9d (maybe next year these two will start playing nice.)  

so long and thanks for all the fish.

3djc, 2018-03-15:
You clearly have an issue that we have never heard off (and there are a LOT of x10 otx downloads); so I would suggest to take things in a more structured way to see if you just go faulty hardware or if there is something software related

sw3Dp, 2018-03-15:
3djc thread tag, you're it.

that was my first panic!  oh no, i got a faulty radio but i've seen this illness before and for the record, there is absolutely no issues with the radio under FrOS command.  i even walked through 3 flash rollercoaster rides to confirm it is a opentx only, conflict with horus.   turned off the internal antenna, ran both and the issue only shows in opentx.  but the big tell is that companion simulator also replicates the stick disfunction, sometimes.  if it's a hardware issue FrOS and my glider don't seem care.

3djc, 2018-03-15:
Well, since companion does not use real stick, or any raido timers and likes, but a completely different system, what you are suggesting is well ... interesting.

So once again, would you be try to be constructive and give us a replicable way to see your issue (wich should be easy since you can replicate on simu), or are you just lookiing for a place to vent, in which case, this is the wrong one

3djc, 2018-03-16:
Fixed as per rocket discussion :)

closed 2018-03-16 by 3djc