opentx/opentx #5777: Unable to check for updates

cperedamas, 2018-03-20:
I am using Companion 2.2.1 on Windows 10 with a Horus X10S.  I have followed all the steps but not matter what program settings I choose, I get "unable to check for updates" both when the program boots up and when I try to download the firmware.

I am using Windows defender and Companion has access through the firewall.  Running companion as admin changes nothing.  I can read the firmware to back it up, so there is communication with the radio.

Below is how the software is configured.

[image]
[image]

3djc, 2018-03-20:
You seem to have an issue that is preventing companion to access the OpenTX servers. The servers are running fine, so it is likely local to your PC, network or ISP, difficult to say from here

cperedamas, 2018-03-20:
Thanks for the fast reply.  I'll try it when I get home.  I am using FIOS.  Have there been any complaints about them blocking your server?

3djc, 2018-03-20:
No, we are not aware of any difficulties getting to our servers, but it is the nature of Internet to sometimes have interconnections that go down for short periods. It could be something like that among other things of similar probabilities

closed 2018-03-20 by cperedamas

cperedamas, 2018-03-21:
It was my FIOS router firewall setting.  Had to be lowered to minimum security.  In case someone else has a similar problem.