opentx/opentx #5781: TX antenna problem! [RTN] on Horus X12S when binding

mdpolasek, 2018-03-22:
I have a Horus X12S flashed to  OpenTX 2.2.1RC99. When I try to bind a new model using the internal antenna I get an "Radio antenna defective" voice error message. The screen shows "TX antenna problem!". 

This looks to me like a software bug, because I don't have any issues with the internal antenna when using FrOS instead of OpenTX.

3djc, 2018-03-23:
In all likelyhood, you really have an antenna issue, as you are the very first to report this.If there was a software bug, we should have had tons of report about it already.

As for FrOS,it could means they are not testing antenna status, or not with the value we use.

Could you go to Long Enter/Statistics/PgDn/PgDn in Analogs screen and let us know the values on the last line (RAS and XJTVER) ?

mdpolasek, 2018-03-26:
The value of RAS flips between 0 and 1, XJTVER is constantly 512. 

I bought the Horus in Switzerland and flashed it with the EUROPE FrSky bootloader. Can this be a reason for this issue?

3djc, 2018-03-27:
No, EU-Lbt is no problem. Did you get those value with internal antenna selected ?

mdpolasek, 2018-03-27:
Yes, these are the values with the internal antenna. If I change it to "Ext + Int" I get slightly different values:
RAS 8, XJTVER is constantly 512.

3djc, 2018-03-27:
Then it is very strange you get this alert, I guess your module gets defective only during bind process, very strange.

martinpolasek, 2018-03-27:
This is indeed strange. The funny thing is that everything works perfectly under FryOS - never had any issues with the antenna. And that I can bind the receiver under OpenTX using the „int + ext“ antenna setting.

Is there a way to debug this issue further?

3djc, 2018-03-27:
The fact that you do not get an error with FrOS does not necessarly mean much, since we do not know if/how this error is handled by FrOS. If you get this with a brand new unprogrammed model, then that can only mena your hardware is somehow defective.

3djc, 2018-03-27:
I may have an idea to diagnose further, do you have a firmware with lua option ? (luac also maybe ?) ?

martinpolasek, 2018-03-27:
I can flush Horus with a firmware with lua option enabled. This is not an issue.

3djc, 2018-03-27:
Then do that, and put the content of the attached file in /SCRIPTS/MIXES, and 'activate' it using custom function screen
[image]

[image]

Once it is done, discover sensors, you should have a new one called DBG. Do a special function that audio it's value every second.

[image]

You should hear it say values between 1 and 5, something like that. Go to your binding process, and let us know what is the value told when binding

[debug.zip](https://github.com/opentx/opentx/files/1851618/debug.zip)

martinpolasek, 2018-03-27:
Thank you very much for your answer and the attached script. Will test it today evening and will let you know the result.

mdpolasek, 2018-03-27:
You managed to run the scripts and it worked immediately. The RAS value was before the binding procedure 0 or 1, but jumped to 96 during the binding procedure. Does this mean that the internal antenna is effectively broken?

schwabe, 2018-03-27:
or the firmware on the internal module just sends nonsense while binding

mdpolasek, 2018-03-27:
You are right! It's indeed an issue with the firmware of the internal module. I downgraded to firmware version to 1.4.04 (from 1.5.03) from there https://www.frsky-rc.com/horus-x12s/ and reflashed OpenTX. And now the issue went away magically :-)

3djc, 2018-03-28:
We are in contact with FrSky on the subject, will update this thread when more info becomes available

3djc, 2018-03-28:
It looks like your radio is fine and there might indeed  be an issue witth the iXJT firmware in the latest revision that sometime raise that alarm when it should not. My understanding is that it will be fixed in next FrOS version (and in the meantime, downgrading like you did is a perfect workaround).

1Sven, 2018-03-28:
I hope it does not affect transmission security. I think it's very important that clarity arises here with FRSKY.

3djc, 2018-03-28:
Why should it ? RAS is hardly usefull anyway (many taranis do not have RAS at all (some X9, all X7,..)

mpjf01, 2018-03-29:
I have a X12S on 1.5.03 and get a bind error message (SWR Warning) on that, as does everyone else I believe. It has been mentioned on RC Groups and advice is given there that is will be fixed in the next release.

3djc, 2018-03-29:
As I said, I have talked with FrSky, they are aware of the issue, and it will be fixed in next FrOS version. Closing this issue as everythiiing has been said

closed 2018-03-29 by 3djc

arnovi, 2018-04-11:
Hi guys,

Just updated Horus ixjt module using latest realease (1.5.04) and the RAS warning has disappeared.
I have not tested in flight but range test result is OK.

3djc, 2018-04-12:
FrSky has confirmed me that FrTX 1.5.04 should fix this issue