opentx/opentx #5783: Q X7 - Expected ver 2.2V0013 error

Skallatorc, 2018-03-23:
Taranis Q X7
Firmware: opentx-x7-lua-en-2.2.1
SD Contents: sdcard-taranis-x7-2.2V0013

Error: Expected ver: 2.2V0013

I purchased this tx a few days ago, updated the firmware and bootloader, downloaded the proper SD Contents file.

I have copied and extracted the zip directly into the SD card, I have deleted, formated, and copied the extracted files from my PC. It has not corrected the alert.

I have verified that the version is correct via the .version file.

3djc, 2018-03-23:
You did something wrong somewhere, difficult for us to say where

closed 2018-03-23 by 3djc

Skallatorc, 2018-03-23:
Wow, it's that simple?

No questions? No did you restart the computer, are you sure the tx is in?
Not in India in a tech support where house by chance?

I have followed the OpenTX website directions, multiple YouTube videos by
skilled people who did the very same steps that fixed theirs, etc

How about some suggestions instead of "sorry but you screwed up and not
worth helping"

3djc, 2018-03-23:
Since this is not a end support site, but a bug tracker, yes, "that simple". For support, you should try user support forum like rcgroups or openrcforum.

Skallatorc, 2018-03-23:
Thank you for the recommendations.

Skallatorc, 2018-03-24:
Just an update, potential bug.

Turns out it wasn't the execution that was the issue. I installed
sdcard-taranis-x7-2.2V0010
and the SDCard folders are listing as they should when editing the
settings, sounds are working but the error still comes up.

The bug seems to be in relation to the OS version and the recommended CD
contents file.