opentx/opentx #5786: Bluetooth trainer issues (X12s)

EmanuelStassar, 2018-03-26:
I have been using Bluetooth trainer option between two X12s and the experience has not been very good. There are a few things I would like to bring up:
-The connection between both transmitters is often tricky to get going. I have to go into the menu and switch bluetooth trainer off and on again to get the other transmitter to discover the signal. Just selecting the correct model on both transmitters has never worked so far.
-If the student walks away to far, the trainer signal is lost, if he walks back, it often takes a very long time before the signal comes back. 
-The quality of the trainer signal is not good. It seems like data is coming in too slowly. Surfaces just don't move smoothly. On a big trainer this is still flyable, but on a miniquad its impossible.

For now I have gone back to using a CPPM receiver on the back of the master radio that is bound to the students radio. The trainer plug input works soo much better. 

I hope the Devs can figure out a solution because it would be a very elegant solution to use Bluetooth.
Oh I am running OpenTX 2.2.1 on a X12s non EU.

3djc, 2018-03-26:
There is not much that can be done about that software wise. The link is managed at hardware level and we cannot do much about it. What you experience is also what we have been hearing so far.

EmanuelStassar, 2018-03-26:
Thats too bad. So that means it also does not function well on FrOS?

3djc, 2018-03-26:
I would assume so, we certainly would like to know too

closed 2018-04-12 by 3djc

kilrah, 2018-04-12:
FrSky have confirmed they only recommend it for "basic slow trainer planes" from their side as well aka with the hardware as it is it's just not very good. 

Seems it comes from the choice of Bluetooth LE hardware which emphasizes low power consumption above all and datarate/range suffers from it.