opentx/opentx #5787: Altitude Autooffset does not work with FrSky G-RX8 vario

Allerhopp, 2018-03-27:
It seems, altitude data comes too fast, to enable autooffset. 
It can easily be verified, if the inbuilt vario is disabled (long press bind button) on startup and enabled after start -> autooffset works in this case.

I remember a similar issue with openXsensor vario, where the dev had to add a delay, to enable OpenTX autooffset. I have told FrSky the context, but whether i could not explain it, and/or they did not understand, or FrOS does not show the issue. It seems, they will not modify their firmware. Therefore a manual telemetry reset is needed after every startup.

Can the OpenTX autooffset routines be modified to prevent this?

3djc, 2018-03-28:
The issue is that none of us does have such a receiver, so it is difficult to replicate/diagnose and/or validate  any fix we may do, so while we will adress this, it might take a little longer than usual

kilrah, 2018-03-28:
Can you validate the receiver isn't sending bogus initial data? No reason something would work wrong if it doesn't...

Allerhopp, 2018-03-28:
Good point. Indeed Autooffset seems to work with the first incoming value, but altitude rises within 1s to 35m in my last try. It seems to be a G-RX8 firmware thing only. But if FrSky does not change the firmware, is it possible, to delay altitude autooffset for a second or two?
As 3djc said, hardware is coming, you can see yourself. Please feel free to close this issue.

Allerhopp, 2018-03-29:
I asked mstrens (openXsensor dev) about the delay, here is his answer:

"In oXs I wait 5 sec before sending Alt.
It is in oXs_ms5611.cpp
You have this line:
 nextAltMillis  =  5000 ;  // in msec; save when Altitude has to be calculated; altitude is available only after some delay in order to get a stable value (less temperature drift)
The delay is quite long but it was to reduce the altitude drift when temperature of sensor increase."

FrSky should fix this imo, but .......

3djc, 2018-03-29:
Once we have the unit, we will assess in details and discuss with FrSky. One way or another, I'm confident a solution will be found

raphaelcoeffic, 2018-04-10:
I think there are some issues with that receiver anyway: it drifts heavily, whereby I've been said that it is half solved with the latest firmware (probably temp related drift). I do wait a couple of minutes before reseting telemetry with that one.

3djc, 2018-04-10:
Still waiting for a unit for complete assessment

SHezinger, 2019-02-26:
I can confirm that the receiver sends it's altitude values before they settle. 

Last year I made a DIY vario for the Graupner/JS Hott telemetry with the same preasure sensor (MS5611) which works pretty good. But to get the auto offset right I had to wait some seconds. 

I will check the temperature drift of the G-RX8. It should not be that bad, because the MS5611 has an internal temperature sensor. With that temperatue value and the give formula in the datasheet you can compensate the drift pretty good. I would say less then 2 meters tops (depending on the time)

3djc, 2019-02-26:
I need too come back on this, g-rx8 alt precision is ok, but less accurate (it drift more) than the high precision vario sensor. Still, should be ok for most usage

SHezinger, 2019-02-26:
They advertise that the G-RX8 has a high precission vario inside:
"Variometer sensor: the measures range is -700m-10000m with the precision of 0.1m (high precision version), support altimeter (the rate range is  +-16.7m/s)"

I assume they have the same preasure sensor build in as in their high pressision Vario module. 
With my experience with the MS5611 I think it shouldn't be a big problem for Frsky to solve the drift issue.

closed 2019-11-23 by stale[bot]