opentx/opentx #5790: New Feature request, Bootloader "Run App" option

MikeBland, 2018-03-29:
I have updated the bootloader for ersky9x to include a "Run App" option. This allows an program ("App") to be loaded into RAM from the SD card and then run. This is useful to easily provide special facilities. I already have one "App", the abiltiy to flash the Multi Protocol module from a file on the SD card.
New facilities may be added simply by writing a new "App", without requiring the main firmware to be updated/released.
The bootloader (and app) source code is at https://github.com/MikeBland/OpenRcBootloader.
On STM processors, apps use RAM from 0x2001000 upwards, an area unused by the bootloader. Currently, apps are restricted to 32K of code (mainly so they fit on the original SKY board thatonly has 48K RAM).
On Atmel processors, the bootloader now uses RAM starting at 0x20008000, leaving the first 32K free so apps may be loaded at 0x20000000.
Apps need to be compiled for the target radio and include a "marker" to indicate the target (e.e. "APPX9D", "APPSKY", "APPX9P" for the plus). The bootloader checks any app is correct for the radio hardware before executing it.

Mike

powercycle, 2018-04-06:
This is so cool Mike.

raphaelcoeffic, 2018-04-22:
@MikeBland thx Mike. For my part, I'd prefer to have the multi-module update directly in the SD card explorer. @schwabe what do you think?

raphaelcoeffic, 2018-04-22:
@MikeBland where is it? I could not find it in the bootloader git.

MikeBland, 2018-04-22:
The changeID app(s) are posted here: http://openrcforums.com/forum/viewtopic.php?f=7&t=4676
I haven't committed the source code to Github yet.
Bootloader and app code is here: https://github.com/MikeBland/OpenRcBootloader
The serial bit-bashing code is in swSerial.cpp, and relies on interrupt priorities being set correctly so the receive serial is the highest priority interrupt with nothing else having the same priority either.

Mike

raphaelcoeffic, 2018-04-22:
@MikeBland could you please upload the source for the ID changer? That would be more convenient ;-)

MikeBland, 2018-04-23:
Source code now uploaded.
An important/useful feature of having these utilities available as loadable apps is they do not rely on having a full release of the firmware, or needing to run "nightlies" in order to have them available.

Mike

kilrah, 2018-04-23:
But they have the huge drawback of having to be built for each target separately including different code for the various hardware and navigation... lots of multiplied work which is really not a fun thing for us to look forward to with the number of current and future supported platforms. When built into the firmware then all the handling is there already.

MikeBland, 2018-04-23:
A certain amount of work to create things, although I have common files for most things and drivers for the different hardware already. In my opinion, these apps don't need much "maintaining", once written and working they shouldn't need changing. They certainly shouldn't be changed/updated when a firmware release is done as they are independant utilities. If built into the firmware, then all "75" would need to be tested as part of a release to make sure other changes haven't "broken" them, which, I would say, is more work than building them separately.
Also, things like the FlashMulti app need a software driver to receive non-inverted serial data over the SPort signal. You said openTx wouldn't support such a driver!

Mike

closed 2019-11-15 by stale[bot]