opentx/opentx #5791: Cumbersome / Slow to get to a BF LUA script on X10/X10s

rhys100, 2018-03-30:
Currently to load the BF lua script on a X10s you need to do the following:
hold down system, click page, click right button, long hold on bf.lua, execute

Thats 5 clicks, and 2 of them are long clicks.

Can is suggest the special scripts can be used to simply click a switch and it loads. Currently it cant load as it cant have UI access.

three other possible options
1. enable long press on pgup.dn like taranis to get to the script.
2. add another option under long press on right button under model select.

rhys100, 2018-03-30:
would it be possible to add a lua script as a main view to replicate what is simialr on the taranis

kilrah, 2018-03-30:
> would it be possible to add a lua script as a main view to replicate what is simialr on the taranis

That is what is planned.

closed 2018-03-30 by kilrah

lshems, 2018-03-30:
It can already be done this way.

You can write a widget that loads the Lua script

rhys100, 2018-03-30:
Go on... Please share your secret

lshems, 2018-03-30:
Oh, sorry. Yes. 

Read my blog on rcgroups.

I'm not saying it is easy, but the tutorial and proof of concept is there.

Thing we really need is button input support for widgets, in whatever form.

rhys100, 2018-03-30:
yea had a look, to hard for now. Will wait for a fix. Agreed re button

lshems, 2018-03-30:
I'll see if I can post a loader script for the bf script on my blog.

lshems, 2018-03-30:
Ok, saw that this was already addressed in the bf github, and the button thing was discussed as well in an issue here.

Just check the Lua bf github.

Or pm me on rcgroups.