opentx/opentx #5794: Special Function Channel Override does not have sufficient granularity to match Spektrum receiver expectations

AlgebraicMirror, 2018-03-31:
On the special functions page, the channel override function doesn't allow sufficient granularity to be able to match the expectations of certain Spektrum receivers/FCs. For example, when using the override function, you can pick values like -107 or -108, but not -107.5. In other screens, such as the output page, you can set limits like -78.2, and the extra decimal place allows for very precise tuning down to the individual microsecond.

Here is an example of why this same level of granularity is needed for function overrides. Horizon Hobby's X-Vert model has an extremely precise requirement for a throttle cutoff switch (and throttle cutoff is 100% required for landing and for safety in the this model, since once armed the props operate in a Betaflight-like "air mode" and will not stop spinning even if the throttle gimbal is completely lowered). You must send the X-Vert a very exact throttle value (equivalent to Spektrum -130), and it must be within +/-1us. This is typically done by assigning a channel 1 override to a switch. However, there is a difference of several microseconds between the override values of -107 and -108 (which are the Frsky X7 approximate equivalent to Spektrum -130), and I found that for my particular transmitter neither of those was close enough to trigger the throttle cut function on the X-Vert flight controller. However, if I modified Channel 1 on the output page so that PPM center was 0.2 instead of 0.0, and then set my throttle cut to -107, that extra .2 variance was enough to get it into range, and the throttle cut functionality started working. 

But this is really an undesirable hack, because the PPM center point did not need modification and should have been able to stay at 0.0. It was the channel override that need modification, but OpenTX did not allow enough granularity to do so, since you can't have decimal points on that page. If manufacturers like Spektrum are going to manufacture products with such tight tolerances that you can't be off by more than 1us, then OpenTX will need to add granularity in places like Channel Override in order to match them.

Here is the RCGroups thread where this example with the X-Vert and the Frsky QX7 is discussed in more detail.

https://www.rcgroups.com/forums/showpost.php?p=39432416&postcount=7508

kilrah, 2018-04-01:
This would be a major change and there are enough other means to do this, several people actually consider this should be done differently to an override...

BTW my X-Vert is happy on -108 using an override.

closed 2018-04-01 by kilrah

AlgebraicMirror, 2018-04-02:
Is there somewhere where I could put this to be considered as a new feature then, perhaps as a request for enhancement?

I'm not sure what you mean by "several people consider override shouldn't be used for that." The OpenTX Companion new model wizard actually asks if you want a throttle cutoff, and then implements it by using channel override. So this is in fact the "official" way OpenTX thinks throttle cutoff should be implemented.

And I'm glad your X-Vert works at -108, but you are proving my point about the tight tolerances. I know another member who's X-Vert works at -108 also, but I can tell you mine worked at neither -107 nor -108. Because the range of values is so narrow, it is hit or miss as to whether a whole number can hit them - it can be thrown off simply by what batch of external module RF chips you got. For some users, such as myself, decimal points are required.

Any chance you would consider re-opening this?

kilrah, 2018-04-02:
> Is there somewhere where I could put this to be considered as a new feature then, perhaps as a request for enhancement?

You did just right here - but that's what comes out of it.

> I'm not sure what you mean by "several people consider override shouldn't be used for that."

Seems you've found the releated issue.

> So this is in fact the "official" way OpenTX thinks throttle cutoff should be implemented.

More that it was the quickest to implement in the wizard, not really that it's the best. A Replace mixer line is arguably better.

> For some users, such as myself, decimal points are required.

And you've got many other places where they're available, so 

> Any chance you would consider re-opening this?

Given the amount of background work needed for something that isn't the best use for it AND the availability of other solutions, most likely no.

AlgebraicMirror, 2018-04-02:
> Seems you've found the releated issue.

But it's not really a related issue. In that other thread, you are asking that the Companion new model wizard not use channel override to perform throttle cutoffs. What I'm asking for is totally different. 

I am requesting that the channel override feature be given decimal level precision like other parts of the product. This has applications beyond the "throttle cut" scenario. I included my throttle cut issue with the X-Vert in my original request because I believe that all feature requests should include sample scenarios explaining ways in which the feature could be beneficial to an end user, but this is not purely about throttle cut - it is about giving the channel override the same level of precision that we have everywhere else in the product, such as the output screens.

I would like to respectfully ask that you re-open this issue for wider consideration, because it is not covered by that other issue and should stand alone. (And I don't expect this to automatically make it into the next version of OpenTX or something, but would like it to be considered and prioritized like other user stories - maybe it makes the cut or maybe not).

> And you've got many other places where they're available, so...

Changing the PPM Center value as I did is a hack that managed to work in this case, but the problem is that it affects more than just the override value that is being sent when the switch is thrown. It changes the entire servo range. When all you want to do is send a very precise override value, you should not have to move the entire servo range to do so. Yes it worked, but they are not equivalent solutions and the PPM Center method has unnecessary side effects that could be alleviated by allowing a decimal point on channel override. 

As for the mixer line solution, when you are trying to send a value outside the range defined on the channel outputs range (in this case, Spektrum -130 where the output range is Spektrum -100 to 100), a mixer line that is factored in before the output limits are calculated is not the best/most natural place to do this. If the value were going to fall within the output limits, a mixer line might be more appropriate. But sometimes, like this case, you need to send a very specific value outside the range, and it's most cleanly and logically done with a channel override.

> given the amount of background work needed for something that isn't the best use for it 

I don't think level of work should be a reason for closing this request. Level of work might be a reason it doesn't make it into the list of features to be implemented in the next version, but I don't think it should be a reason for completely removing it from consideration, particularly before a wider audience gets a chance to see and comment on it - maybe another developer would see it, want it, and like to take up the cause.

Also, I will admit, I don't have the OpenTX toolchain downloaded and haven't gone through the source code. But, it seems like this might not be that much work? I use a Frsky QX7, which has probably the lowest resolution display of any of these transmitters, and yet it does have room to display a decimal place. And certainly Companion could handle the display of an extra decimal place. Internally, the code is capable of decimal level precision since they are used on the outputs page, so this seems like a display level change that might not be that bad - it might be limited to a UI change.

> BTW my X-Vert is happy on -108 using an override.

Also, another user on RC Groups just ran into this issue with the X-Vert, in which no whole number value was precise enough to trigger a throttle cut, so there are probably quite a few people affected by this (probably varies by RF chip batch), which shows why this could be worthwhile:
https://www.rcgroups.com/forums/showpost.php?p=39439510&postcount=7539

But again, I don't want to get distracted by the "throttle cut" scenario. It is an example scenario - the actual request is to have the channel override page be consistent with the rest of the product by allowing decimal level precision - which could be used in other ways.

schwabe, 2018-04-02:
You have not told us what external module you are using for DSMX. Also most BNF remote are not precise. So I am kind of confused that this precision should be required. Also the manual talks about throttle cut on the 6th channel.

For MULTI:

By default MULTI cannot go more than -/+125% (OpenTX values) for Spektrum. And 100% OpenTX is not 100% spektrum either:

By default:
1100..1900us @100% and 1000..2000us @125%.

(you can change #define DSM_MAX_THROW to achieve different values)

In the post you linked you talked about trying to achieve 980µs which is outside the normal range.

Your problem might be that you have different multi versions on your modules which interpret 108 different and  you end up with different output values. (one more true to 100% OpenTX => 100% DSMX and the other more true to OpenTX µs => DSMX µs)

For your setup I would use a replace line instead of a channel override as @kilrah to double check if the precision needed is really the problem here.

Also assuming that you indeed using multi  or a DIY DSMX module from a Spektrum doner. All values are digital if you use channel override down to the serial stream to the module and the over the air transmission. Which means that two OpenTX version will produce **exactly** the same output at the receiver. If the reciever is indeed as precise it should either always work on all models or on none. (And original Spektrum remotes would have the same problem).

AlgebraicMirror, 2018-04-03:
> You have not told us what external module you are using for DSMX. 
My apologies. It is a Spektrum DX4e RF chip, using the [plans found here](http://johnprikkel.blogspot.com/2014/07/jr-dsmx.html). It is home built, but the design is super common as you can find people selling this exact custom circuit board on RCGroups, eBay, etc all the time.

> Also most BNF remote are not precise.
I can't speak to that, since there are many and I only have a few. I would assume that is probably true particularly when using a traditional receiver that doesn't have a lot of complex programming and just uses the value directly, although Spektrum seems to be doing more and more of these heavily programmed models like X-Flight and Convergence.

> So I am kind of confused that this precision should be required.
As an engineer, I personally think it's crazy to have such tight tolerance, because I wouldn't be surprised if a slightly different batch of their own radios have different enough silicon that it causes them to be off by more than a microsecond, giving them issues. Nevertheless, that's the route Spektrum has chosen to go, and it's been proven now by multiple users that they are did in fact make the design this tight.

> Also the manual talks about throttle cut on the 6th channel.

Throttle cut isn't on channel 6. The motor arm switch is on channel 6 (described as Switch I on page 6 of the manual), and it doesn't replace throttle cut (described as Switch H on page 6). Unfortunately, the directions lack detail when it comes to describing the cut switch, but it's because they assume you are using a Spektrum transmitter, and all Spektrum transmitters are programmed to have throttle cut send an override signal on Channel 1 of -130. Every one of their transmitters sets up every new model this way. And the X-Vert is no different. I used the DXe programming cable to look at the included programming on the DXe that is in my X-Vert RTF package, and it is a channel 1 override of -130 that does throttle cut, same as every other model.

Now, with other BNF models with more basic receivers, tolerances are not always so tight. I basic receiver will interpret anything under -100 as a "zero" for the purposes of throttle position, so you can cut throttle with a wider range of values. But with these highly programmed receiver/flight controllers, apparently they are looking for super tight precision. This definitely includes X-Vert, but could also include other specialized models like maybe Convergence or some of the quads, etc, but I haven't checked since I don't own them all.

Here is why throttle cut is so important on X-Vert, and especially why I think we may see this problem on some of the new or soon to be released advanced Spektrum quads that have an equivalent to Betaflight's "air mode". The X-Vert basically does "air mode" when in vertical takeoff and landing mode. When you arm the motors, they start to spin, and they cannot be turned off even if you drop the throttle to zero on the gimbals. This is done intentionally, because differential thrust from the motors is used to control direction, which is the same as quads. In quads or this plane, if you cut the motor, you also lose the ability to control direction which can result in a crash. To prevent that and give you guaranteed yaw, pitch, and roll authority during descent, they always spin at a certain minimum level on the X-Vert, just like with Betaflight airmode, and probably newer Spektrum quads are doing this.

The only way to instantly cut your motors is with the throttle cut switch. If you try to use the channel 6 motor arm/disarm switch, it's not instantaneous - you have to hold it down for a couple second before it will cut the motors, which is another part of the design. As a result, not having a throttle cut switch while flying this model is a safety risk because you can't get the props stopped if you hit someone or something. Also, both the X-Vert manual and everyone on RC Groups state that the throttle cut switch must be used in landing, and throttles must be cut right as the craft is touching down. Otherwise, the continuously spinning motors make it too floaty, and it will tip/crash/skip across the ground, rather than set down solidly.

So for the X-Vert, throttle cut is absolutely critical for safety and for landings, and unfortunately, the tolerances are as crazy tight as I've described them. Again, I can't say how many planes/quads are affected, but if Spektrum thought this was an OK idea to have such tight tolerances here, it's probably not the only place it has/will happen. So that's one reason why having more channel override precision would be useful going forward, but it may not be the only scenario where it could be useful.

> In the post you linked you talked about trying to achieve 980µs which is outside the normal range.

It's not outside the normal range for Spektrum. Just look at a DXe. It has a throttle cut switch on it, and that is programmed to be a channel 1 override of -130 for every new model you create. I think most current Spektrum radios do it the same way - it's just part of their design to use -130, which is why -130 is essentially part of their "normal range". My DX4e chip has no problem going outside this range via my external module - just turn on E.Limits, and you can do easily do it - it's only a matter of precision that is the issue. People are doing it using OrangeRX external modules, etc, in RC Groups.

[... cut ...]
One other thing we've noticed is that OpenTX actually isn't correct when it comes to output displayed in microseconds anyway, at least not when it comes to Spektrum and Spektrum compatible models. Mathematically, to hit 980us, the value in OpenTX should be around -102 (and OpenTX agrees and displays something close to 980us on the outputs page when you flip an override switch set to -102). However, this is not what a Spektrum receiver sees coming out. It sees a much different value. Only when you get to around -107 to -108, which OpenTX display as about ~950us, will a Spektrum receiver record a value of 980us. So there is that issue as well.

kilrah, 2018-04-03:
> Which specific way are you suggesting?

Replace mixer line, and fine adjust with subtrim or end points if it happens to be necessary.

> So it can't stay open as a "request for enhancement" that may be picked up later by someone?

Becasue 1) through 5 years of experience the chances of someone picking this up and doing all that's needed correctly are close to 0 so it'd just be clutter, and 2) we still decide on the project's direction and are likely to refuse stuff that is deemed unnecessary based on the added maintenance/support burden.

No problem for any other dev to comment and/or pick it up. As mentioned it's a major change that affects data format so would require conversion routines between OpenTX versions, handling of conversions between radio types, has to wait for many months for the next major version etc for what if you take a step back is pretty much useless.

schwabe, 2018-04-03:
DSM2 has only 1024 steps. So (2096-904)/1024 ~=  1,1. So precision in sub µs is out of the question from a pure protocol standpoint.

But I agree with with @kilrah Your problem is a corner case and there is a very good solution to your problem. And changing the precision for all other cases is not worth the trouble

AlgebraicMirror, 2018-04-04:
I will try what @kilrah suggests as soon as I can to see if it actually works. My X-Vert is currently damaged and I am waiting for some repair parts to be delivered before I can test this again.

However, there still seem to be quite a few misconceptions about what I'm asking for and what this issue is:

> DSM2 has only 1024 steps. So (2096-904)/1024 ~= 1,1.

DSM2 is not involved in this at all. The X-Vert has a DSMX receiver, and the Spektrum DX4e chip I'm using in the external module is also DSMX. And DSMX does claim 2048 bits if that is what you are referring to.

> So precision in sub µs is out of the question from a pure protocol standpoint.

I'm not asking for sub µs precision. I'm asking to simply have single µs precision, which OpenTX isn't providing for the channel override feature. According to the output page on OpenTX, the difference between -107 and -108 is 5 µs. This is why some users are unable to hit the Spektrum cutoff value of 980µs +/- 1µs. When there's a gap of 5 µs in between the values OpenTX makes available, and you can't go any tighter, you can easily miss that. That's not a particularly good level of resolution.