opentx/opentx #5804: Bug in BattCheck Widget

Gewe99, 2018-04-05:
local function getCellPercent(cellValue)
 --## Data gathered from commercial lipo sensors
 local myArrayPercentList = ...
 {{3, 0}, {3.093......
 local result = 100 -- -> Add this line! Otherwise error when starting the Widget while one cellvoltage is greater than 4,2V  (returning a Nil value...)
 for i, v in ipairs( myArrayPercentList ) do
 if v[ 1 ] >= cellValue then
 result = v[ 2 ]
break
 end
 end
 return result
 end

[main.zip](https://github.com/opentx/opentx/files/1879609/main.zip)

Please place the code in the sdcard-horus-2.2V00xx.zip

Regards Dirk

3djc, 2018-04-05:
Are you using high voltage lipo ?  if not you should have more than 4.2, if yes, then the table values are likely wrong

Gewe99, 2018-04-05:
This happen sometimes if I connect a lipo very fast after charging to the model.

The initialisation of the result prevents that error...

3djc, 2018-04-05:
Yes, but it makes me think that I also need to check for highers values to prevent display of incorrect stats for HV lipo

offer-shmuely, 2019-05-13:
fixed as pert of batt-check updates on pull request #6412
can close this issue after accepting the pull request

offer-shmuely, 2019-12-06:
please closed this bug (fixed in #6412)

closed 2019-12-06 by bsongis