opentx/opentx #5818: Horus Lua Widget strange behaviour: options, vars, function memory

zenuavsolutions, 2018-04-11:
have a strange behavior of my Lua widgets. At first I used the same widget in different models but with different options. This had the consequence that in model 1 the options of model 2 showed effect. I rewrote the widgets and also used different names for the options. Now the Horus can not remember which widgets is set on the model and I get after switching on a blank screen. I put the widgets back in the telemetry menu and they work. I turn off and on the Horus and again they are gone ...

I tried the following:
Widgets rewritten (all variables and functions local) -> no change
Models deleted and recreated -> no change
FrOS and again OpenTX geflsht -> no change

lshems, 2018-04-12:
Hi,

Make sure the folder name of the widget is not to long, otherwise settings are not saved. I think that is part of your problem.

Settings between models is strange, but could result from having the code not seperating environments properly. 

Widgets are tricky, because the code is  loaded once, and run as instance for each defined widget

lshems, 2018-04-12:
If you want, you can upload your widget on the repository I'm setting up.

I'll check it and make suggestions if neccesary.

Repository.justfly.solutions.

You can create a login to upload.

lshems, 2018-04-12:
Jochen, I think you used an incorrect email. Please pm me at rcgroups. L shems

zenuavsolutions, 2018-04-12:
oh, maybe a typo, can't find you at RCGroup, i'm there as "webmotions" or delete the user "joa" and i try again...

zenuavsolutions, 2018-04-12:
ok, i have a lot of variables, so i have to define them all in the create function?
But does this also explain that widgets and there settings are no longer saved?

raphaelcoeffic, 2018-04-12:
I did not look at your code very intensively, but yes, that could happen.

lshems, 2018-04-13:
Ok, I have no good internet right now. 
Only on my phone. 

But I've posted some questions on your github. Let's discuss there. 

I don't think you need to rewrite a lot. And no, not all variables need to go into the create part.

closed 2018-04-13 by raphaelcoeffic

zenuavsolutions, 2018-04-14:
Thanks to some valuable tips, the script now works again. You never stop learning :-)
1. Widgets names can have a maximum of 10 characters
2. Variables and functions should not be global in the Lua environment
3. It is best to pass all variables via the create () object

Thanks @all !

raphaelcoeffic, 2018-04-14:
Could you please add your findings to the documentation?

lshems, 2018-04-14:
I think they where already there in the Widgets part of LUA 2.2 guide

'Widgets are located on the SD card, each in their specific folder /WIDGETS/<name>/main.lua (name must be in 8 characters or less).'
'widget name (name must be a string of 10 characters or less)'

'in the example given, you can see that no global variables or functions are needed to operate the widget.'
'variables that are used throughout the widget, can best be declared inside the create function as local variables'
'those local variablkes can then be passed through to the other functions as an element of the widget array that is returned'

raphaelcoeffic, 2018-04-14:
Right, but there is a fine line between CAN and SHOULD ;-)

pd5dj, 2019-09-12:
Question about global variables.

I have a widget that I have running on multiple pages in a model. in other sizes.
I want this widget to set a Gvar.

Do all the "same"  widgets in the model set the Gvar in the background? or only the one that is visible?

lshems, 2019-09-13:
Unless you call background from refresh

lshems, 2019-09-13:
Post the script here or on my rcgroups blog with a short description of the intended workings and I'll look at it

pd5dj, 2019-09-16:
> Post the script here or on my rcgroups blog with a short description of the intended workings and I'll look at it

Will do later on, tnx

Hobby4life, 2019-09-17:
Changed account.