opentx/opentx #5821: Companion: the Logical switch list does not self refresh properly

vian31, 2018-04-12:
Just starting programming with 2.2.1 version and I found that in Companion, while setting up severals LS, the list on parameter V1/V2 and AND sup does not self refresh with the whole list of already EXISTING logical switches. Then not able to select them. To do it we must save the file, close and re-open it, which is boring indeed. See example attached. My profile is for QX7s but it looks the same with X9D (not tried all...)
In the radio the whole list from LS1 to LS64 is available, then no pb. It seems to me that the hidden of not used LS in Companion is new from this version. Am I right ? In anther hand this is a good improvement.

[image]
Vianney

3djc, 2018-04-12:
Known and fixed a while ago, will be in 2.2.2

closed 2018-04-12 by 3djc

vian31, 2018-04-12:
Yep, very good. I was also a bit suprised to be the first hilighting that.
Thanks
Vianney