opentx/opentx #5824: 9XRPRO sound voice issue with timer

blairfancy, 2018-04-12:
Hi,
I put 2.2.1 fw on 9XRpro radio and i noticed that timer voice sound does not work properly.

i set 8min timer running with throttle tick minutes and voice. i heard 7 min, 6min but after that no sound is reported.

kilrah, 2018-04-12:
Only those or all sounds? Have you properly loaded SD card contents/voice pack for 2.2?

blairfancy, 2018-04-13:
all sound have gone even playsound function.

blairfancy, 2018-04-13:
Yes installed voice pack for 2.2

kilrah, 2018-04-13:
The it's not an "issue with timer". Try to format your SD card and reload the files, if it doesn't help try replacing it altogether.

blairfancy, 2018-04-16:
Hi tested all your trick but it does not help. Is there any specification of SD card? speed , partition type, size?

blairfancy, 2018-04-16:
Definitely not a SD card issue because even without it beep sometime is not played

Bogdanu2018, 2018-04-27:
did you fix it ? I have similar issue, radio alerts and sounds stop working while using the radio. When I turn it on everything is fine all sounds good but after few minutes the sounds and alerts stops working... I replaced SD and all files

blairfancy, 2018-04-28:
Unfortunately not resolved and i ve no skills to debug that by myself.Nevertheless i ve toolchain ready for compiling opentx FW. I dont know if there is any option to add some sort of debug or log to have better view of this problem?

njozsef, 2018-04-29:
I have a similar problem.  radio fw ver 2.2.1

dman2552, 2018-05-12:
I am also on 2.2.1 with latest voice files and have the sound issue.  It doesn't appear to be specifically timer related and is very random.  I can have sound stop working at 30+ minutes or sometimes after only a few minutes.  I haven't found any stick or switch use that will consistently reproduce the problem.  It just seems to drop out at random.