opentx/opentx #5829: Sync TX time with Companion

raphaelcoeffic, 2018-04-14:
As we have a new mass storage device displaying certain properties of the bootloader, we could use that to read and possibly write the TX timestamp. Background is basically #5735. It has been shown that SD card sync requires the timestamps to be synched.

However, this needs to be deactivated when GPS time sync is preferred (X12S setting).

raphaelcoeffic, 2018-04-14:
@schwabe Hey! Do you think that would possible? Do you know if operating systems do somehow cache the timestamp seen on mass storage? Or could we always read the current TX timestamp by just checking the file date? When writing to the file, I guess it will automatically take the timestamp from the computer, right?

mpaperno, 2018-04-14:
Most storage devices have no concept of (real) time. I don't think USB MSD has any API for time sync.  Of course you could always check timestamp on some file, but what does that really tell you?  

If you write a file via USB/MSD then the timestamp will come from the computer.  And radio doesn't have access to SD card while MSD is active.

raphaelcoeffic, 2018-04-14:
@mpaperno that's not really true, please have a look at the implementation for the "Taranis" drive. We do generate every single bit there on the fly.

raphaelcoeffic, 2018-04-14:
The only thing I'm not quite sure about is the opsys caching...

bsongis, 2018-04-14:
Why not use USB CDC for that?

raphaelcoeffic, 2018-04-14:
@bsongis is that ON at the same time as mass storage? Did I miss something?

raphaelcoeffic, 2018-04-15:
@bsongis ok, mass storage does not really work for that purpose, as the OS is caching the file, no way around that. Is that somehow possible to enable USB CDC together with mass storage / joystick?

closed 2019-11-15 by stale[bot]

naiba4, 2022-07-18:
sync-tx:6de804e2f2a096272123b21b7856367d177c9d754ec844b6fae39b29915133b6

naiba4, 2022-07-18:
Lütfen yardımcı olun