opentx/opentx #5835: Horus X10s Module Bay Issue

Manoo430, 2018-04-19:
Hi;

Recently I bought Horus X10S, I flashed it to OpenTx 2.2.1. When I tried to use my EzUHF external module I realized that the module is not powering up. 

Looking at attached photo which is showing the JR module connectors of the Horus, I did below tests on both FrOS and OpenTx operating systems on Horus X10S:

[image]

1. In FrOS, the module powers up and everything works normal. The voltage between Vmain and GND measured 6V.
2. In OpenTx, the module doesn't power up! The voltage between VMAIN and GND is almost Zero, the Voltage between VMAIN and S.P is 6V. I switched my negative wire from GND to S.P but when I connect the module voltage drops to 3V from 6V and it cannot power up the module.
3. I used an external battery and it worked good, I had signal too.

I have no programing skills, but I feel OpenTx messes up with the voltage regulators at module bay and also at trainer port ( I had similar issue with trainer port when I was trying to use head tracker - the negative wire was switched to other pin when on OpenTx, I fixed it by shorting 2nd and 3rd rings of my cable connector).

Need assist for this issue.

Cheers

Manoo

3djc, 2018-04-19:
Vmain is turned on when you declare a ext module in the model settings

Manoo430, 2018-04-19:
Of course! this is a very basic thing! during all the tests, my external module setting was on PPM, means ON.

Manoo430, 2018-04-19:
Plus, as I said ...everything works normal when the operating system in FrOS....problem happens when I flash it to OpenTx.

Manoo430, 2018-04-19:
This shows OpenTx is not fully compatible with Horus X10 hardware.

kilrah, 2018-04-19:
First time we have a report of that, and it would be surprising for you to be the first person to try and use an external module on an X10...

Massild, 2018-04-19:
> I guess so, couldn't find any report abt it! Hope OpenTx can fix the bug.

I have 3 external modules that all work with openTX.  I am running 2.2.1 on
a Horus X10S.  Maybe the S makes a difference?

raphaelcoeffic, 2018-04-20:
Just looked it up: EzUHF states 9-12v as input voltage. Could that be the issue?

3djc, 2018-04-20:
Maybe he tried FrOS when battery was more charged and by luck it worked there ?

Manoo430, 2018-04-20:
No I tried it 2 times...on FrOS it is ok but on OpenTx not!! very confusing! same wiring, same hardware!

raphaelcoeffic, 2018-04-20:
@Manoo430 could you please upload your model file?

raphaelcoeffic, 2018-04-20:
And try with a brand new model having only the PPM module enabled?

Manoo430, 2018-04-21:
Hi, Even with a brand new model with PPM enabled on external module I have same issue. will upload the model file when I'm home. Tnx

raphaelcoeffic, 2018-04-21:
Just to be 100% sure:
- PPM output works
- but power output is OFF (~0V)
- exact same configuration works without hardware changes with FrOS

Is that correct?

Manoo430, 2018-04-21:
Hi, yes PPM works, as you remember i mentioned that when I use an external battery, system works.
Also on FrOS i have exact hardware with no changes and it works. Are these two operations systems using different power regulator configurations?

Manoo430, 2018-04-21:
it has become a riddle to me!!!!

raphaelcoeffic, 2018-04-21:
@Manoo430 no, there is only a MOSFET to be switched ON or OFF, so basically nothing CAN be different here between FrOS and OpenTX. Do you have the X10S with the battery charger integrated? The only thing I could possibly think of is that something has changed here as well (hard to believe, but well).

Manoo430, 2018-04-21:
its the latest version with built in charger (integrated).

Manoo430, 2018-04-21:
the question is, if its a hardware issue, why does it work on FrOS??!!

Manoo430, 2018-04-21:
it has warranty, but i don't like to return it...its a unique case! ;) ;)

raphaelcoeffic, 2018-04-21:
I was just thinking, maybe something has changed on the PCB, probably nothing defect.

raphaelcoeffic, 2019-08-17:
@Manoo430 is this is issue still the same for you?

closed 2020-05-13 by stale[bot]