opentx/opentx #5836: windows 10 driver problem

jmckeown5524, 2018-04-20:
As per technical note by Jet Flyer 7/3/2015
RE: Taranis X9D (old but never used)
Companion 2.1.6
Opentx 2.1.X
After flashing 
Message "unable to read radio"
ck'd tx; firmware version didn't change
? usb to tx handshake.
installed Zadig v 3.1, don't understand the options.
Or perhaps there are other areas that I should explore.
thank you 
[image]
[image]

3djc, 2018-04-20:
We are sorry but this place is a bug tracker, not one for technical support. Please go to places like rcgroup or openrcforums

closed 2018-04-20 by 3djc

jmckeown5524, 2018-04-20:
Having moved along a bit with Zadig v 2.3 I loaded a driver "STM32 boot loader" and was able to load opentx 2.1.9 to the X9D Taranis with no screen messages.  Unfortunately the TX read something like " firmware error" and the version number of the firmware in the TX didn't verify the flash.  Now the usb mouse is flying all over the place.  So I'm a little closer, but I need a Taranis driver specifically for windows 10 and probably a bootloader that is compatible with this exercise.  Thank you if you can advise me on what to do next.  Until tomorrow. Jim