opentx/opentx #5840: 5.8GHz Video interference w/R-XSR & Taranis X9DP

Gozzz, 2018-04-22:
I've have this issue with 3 of my quadcopters. On all they are using R-XSR w/f.port firmware version 180228, Taranis X9DP running 2.2.1. One setup runs TBS Race HV vtx, the other runs a Eachine VTX03, and the last running a TBS unify pro nano 5G8.

I get rolling video interference if the radio and quad are powered up too close, at this time I also get the telemetry lost/recovered message (a common problem of swamping out the RX it seems). If I power off the Taranis the video goes clear. I can then power the TX back on and the video stays clear unless I get it close enough to swamp out the RX again.

I know this may not truly be an opentx issue (maybe R-XSR firmware issue with handling the swamping close proximity?) but I was hoping to be able to see if anyone else had the problem or could recreate it and troubleshoot further? 

FWIW all the builds are pod style with the RX, VTX, FC in close proximity to each other. They all fly and perform fine as long as the RX isn't swamped out.

raphaelcoeffic, 2018-04-23:
Sorry, but this really is not the place for this kind of discussions. Please use RCG, or maybe some other forum.

closed 2018-04-23 by raphaelcoeffic

Gozzz, 2018-04-23:
Thanks @raphaelcoeffic, I'm going to try Frsky support.