opentx/opentx #5842: Transmitter should ignore messages not listed on telemetry tab.

marbalon, 2018-04-24:
I'm using FrSky receivers just to retransmit RC signal to my tracker. RC data is transmitted using SBUS but my tracker transmit back telemetry using SPORT - information from done like battery, GPS, etc. but also with RSSI, and here is a problem. When I have a connection to my plane I receive telemetry and everything is ok, but when I lost telemetry send RSSI = 0 from tracker to SPORT receiver and this generates me "Telemetry lost" but FrSky receiver send it own telemetry with RSSI and Taranis report "Telemetry received".

My tracker generates SPORT frames with ID other than FrSky receiver, and I remove all sensors from telemetry tab but with ID 0x25 but the problem still exists.

So IHMO software should ignore sensors not listed on telemetry TAB and if you have two RSSI sources or two battery information you can select what to ignore.

raphaelcoeffic, 2018-04-24:
Do I understand it correctly that you have 2 receivers bound to the same module, and each of them is sending telemetry?

marbalon, 2018-04-24:
No,
I have one LRS transmitter on my tracker connected to taranis using FrSky receiver (SBUS + Sport). So telemetry received in LRS is transmitted to FrSky receiver via S.Port. So taranis receive telemetry generated by FrSky module and telemetry generated but LRS.

PLANE(LRS)   <air>   TX(LRS)  <SBUS + S.Port> FrSky  receiver   <air> Taranis.

So when I select discover sensors in my Taranis I have a lot of sensors from LRS and three sensors from FrSky receiver (of course with a different ID). I delete all sensors with FrSky ID, but still, Taranis receive it internally and mix information about telemetry lost/recover.

raphaelcoeffic, 2018-04-24:
ok, so basically you have some sensor (your LRS module) connected to your Frsky receiver that sends the same type of packets as the receiver (RSSI), only differing by the Sensor ID, right? Do you happen to have a trace of this? It would be interesting to see what is seen by your Taranis.

marbalon, 2018-04-24:
Of course, all sensors from LRS use different ID than FrSky.

kilrah, 2018-04-24:
Then there should be no problem. More detail is needed then.

marbalon, 2018-04-24:
I tested on two different FrSky receivers with the same results X4R and XSR.

kilrah, 2018-04-24:
I mean, more than one powered simultaneously.
If only one FrSky receiver is powered and your LRS uses other IDs then there is nothing that would casue this apart from an actual loss of telemetry feed.

That can happen if the radio is too close to the receiver which is very common knowledge, you didn't mention how far you spaced your stuff out.

raphaelcoeffic, 2018-04-25:
I think that LRS is using a different sensor ID, but the same value ID, is that correct?

marbalon, 2018-04-25:
@raphaelcoeffic 
Maybe now it will be more clear, please see attached foto. LRS use different ID (4), than FrSky (25). and like you see Taranis have two RSSI sources. Then I delete sensors with ID 25, so Taranis should wait only for ID=4.

And now scenario looks like here:
- when LRS plane is ON LRS generate telemetry and it goes to taranis and everything is working
- when I turn OFF model, LRS lost telemetry and generate only RSSI frame with value 0
- Taranis receive this frame and recognize it as telemetry lost, but still, FrSky receiver sends RSSI with ID=25 and Taranis generate the message "telemetry received", and again LRS send RSSI=0, and telemetry lost, and again, again...
Now it is more clear ?

[image]

kilrah, 2018-04-25:
Yes, it confirms the issue is your LRS using the same Data ID as FrSky as I first suggested.

marbalon, 2018-04-25:
Yes LRS transmit RSSI as 0xf101 just like FrSky, so you suggest use different to send RSSI ? But how openTx recognize RSSI source ? using Data ID ? 

But when I change Data ID openTx will not know that it is RSSI and will not inform me when telemetry from LRS is lost. IHMO better option is ignore sensors not listed on telemetry TAB.

kilrah, 2018-04-25:
Yes, use a different ID. You can discover that, select the proper unit, then set up your own alarms on it using logical switches and special functions. You get the benefit of being able to select different alarms/voices to distinguish both then.

Everything that isn't in the list is already ignored... EXCEPT RSSI which is a special case becasue it's used to signal telemetry presence and to feed the internal warnings even if the user hasn't discovered sensors.

marbalon, 2018-04-25:
But maybe it is a good idea to select RSSI source also like voltage source?

raphaelcoeffic, 2018-04-25:
Seems reasonable to me. But what happens if more than one sends RSSI, just like in your initial case? Which one should be chosen? All of them?

marbalon, 2018-04-25:
Yes, all of them. If user configure 2 sources, he probably know what is he doing ;-)

I've just browsed the code and the simplest way is define function in telemetry_sensors.cpp like

isValidRssiInstance(uint8_t instance) with results:
true - if sensor not found or founded with the same instance
false - if there is a sensor with different instance only

And then use it in sportProcessTelemetryPacket(). I can even code it but just let me know if this is correct solution.

raphaelcoeffic, 2018-04-25:
Cool, just create a pull request and we can have a look at it!

marbalon, 2018-04-25:
OK, pool request created. Tested and working as expected ;-)

closed 2018-06-16 by kilrah